RavenSystem/esp-homekit-devices #1231: Inching Time for Thermostat Devices

macjl, 2020-11-27:
Hello,

Is the inching time supported for Thermostat Devices?
I have heaters in bathroom that don't need to stay active for more than 1 hour. So to not forget them, I would like to add a inching time to bring back the heater to Off state 1 hour after start.
As I didn't find the information in wiki, I'm asking if it' already supported?
Or do you think it could be added in a future release?

macjl, 2020-12-08:
Hello @RavenSystem

Do you know if inching time for heater cooler could be implemented?

RavenSystem, 2020-12-08:
You must use a temporized switch and internal notifications actions.

macjl, 2020-12-08:
Oh! I didn't thought about it. I'll make a JSON like this tonight.
Thank you!

macjl, 2020-12-08:
Done ! For those who want similar behaviour, or example, that's what I've done : 

Old Json : 
`{"c":{"z":0,"o":3,"l":13,"i":1},"a":[{"t":21,"0":{"r":[{"g":12,"v":1}]},"1":{"a":0},"2":{"a":0},"3":{"r":[{"g":12,"v":0}]},"4":{"a":0},"5":{"a":0},"g":3,"n":3,"d":1},{"1":{"s":[{"a":1}]}}]}`

New Json : 
`{"c":{"z":0,"o":3,"l":13,"i":1},"a":[{"t":21,"0":{"r":[{"g":12,"v":1}]},"1":{"a":0},"2":{"a":0},"3":{"r":[{"g":12,"v":0}],"m":[{"g":3,"v":1}]},"4":{"a":0},"5":{"a":0},"g":3,"n":3,"d":1},{"1":{"s":[{"a":1}]}},{"i":3600,"h":0,"0":{"m":[{"g":1,"v":0.02}]}}]}`

Thank you for your help @RavenSystem !

closed 2020-12-08 by macjl