RavenSystem/esp-homekit-devices #1232: AOFO 4AC+4USB Tuya Power Strip

TheReiner, 2020-11-28:
Hi,

i don't find the configuration for this product, can you help me ?

There is a template for Tasmota but not for this firmware.

[https://templates.blakadder.com/aofo_4AC+4USB_tuya.html](url)

hillaliy, 2020-12-12:
{"c":{"l":1,"b":[{"g":3,"t":5}]},
"a":[{"t":2,"s":5,"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[{"g":3}]},
{"t":2,"s":5,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":3}]},
{"t":2,"s":5,"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":3}]},
{"t":2,"s":5,"0":{"r":[{"g":13}]},"1":{"r":[{"g":13,"v":1}]},"b":[{"g":3}]},
{"t":2,"s":5,"0":{"r":[{"g":14}]},"1":{"r":[{"g":14,"v":1}]},"b":[{"g":3}]}]}

TheReiner, 2020-12-12:
It's a similar product but it's not... 
Mine (AOFO 4AC+4USB Power Strip)  is with MCU ...
i try this configuration (Geekbes 4AC+4USB Power Strip) before post this issue.

tonysprenk, 2020-12-12:
If it has a secondary mcu you need to figure out the commands used for the communication between the esp chip and the mcu. In the wiki is a chapter on configuring json with serial commands. On the tasmota wiki is info on the tuya protocol.

closed 2021-01-12 by stale[bot]