RavenSystem/esp-homekit-devices #1235: Fujitsu a/c

pasparakis, 2020-11-28:
Hi @RavenSystem thank you for your amazing work..
Can you help me with my device? 
I made one project with semis d1 mini, first flashing fulhaaboot.bin

and then I used from esp database this json:

{"c":{"t":3},"a":[{"t":25,"w":3,"g":14,"h":-3,"m":19,"x":28,"ds":0.5,"df":0.5,"p":"H0C#AF0:AFB)AF","0":{"i":[{"c":"bAaAcBcBmAgAdAfAaF"}]},"1":{"i":[{"c":"bAaAcBcBmAgAdHbAdAcBbAdAaAbAgAzhAcBe"}]},"2":{"i":[{"c":"bAaAcBcBmAgAdHbAdAcBbAbBaAaAiAzhAbAcAaAa"}]},"3":{"i":[{"c":"bAaAcBcBmAgAdHbAdAcBbAdAaAbAeBziAbCe"}]},"4":{"i":[{"c":"bAaAcBcBmAgAdHbAdAcBbAbBaAaAgBziAcAbAaAa"}]},"8":{"i":[{"c":"bAaAcBcBmAgAdHbAdAcBbAbAbAcAgAzhAcDaAa"}]},"9":{"i":[{"c":"bAaAcBcBmAgAdHbAdAcBbAcBaBiAzhAbAbE"}]},"10":{"a":1},"11":{"a":2},"5":{"m":[{"g":1,"v":0.02}]}}]}
For Fujitsu HVAC Ir Remote Thermostat and Humidity sensor

Gpio pins : 3: IrDA LED, 14: DHT22 sensor..

everything works fine except ir commands .. works only (off)command

I think json is for different remote control (H0C#AF0:AFB)AF)

my own remote control is (AR-RAH2E)

can I help me to use the right code in json and it will work in the end...

hillaliy, 2020-12-03:
https://github.com/maximkulkin/esp-ir-thermostat
maybe it's can help you

pasparakis, 2020-12-03:
Thank you fried

closed 2020-12-28 by pasparakis