RavenSystem/esp-homekit-devices #1236: shelly 1L

miamilabs, 2020-11-28:
Hey Community,

i am successfully using this software for many of my shellys, now i did order shelly 1L and would like to flash it as well.
Will this work on the 1L? Because i can not find a JSON in the supported devices.

Thank you

closed 2020-12-29 by stale[bot]

mrmaximas, 2021-01-18:
> Hey Community,
> 
> i am successfully using this software for many of my shellys, now i did order shelly 1L and would like to flash it as well.
> Will this work on the 1L? Because i can not find a JSON in the supported devices.
> 
> Thank you

did your find proper json config?

miamilabs, 2021-01-19:
> > Hey Community,
> > i am successfully using this software for many of my shellys, now i did order shelly 1L and would like to flash it as well.
> > Will this work on the 1L? Because i can not find a JSON in the supported devices.
> > Thank you
> 
> did your find proper json config?

No i did not =(
To not lie, i am comfortable to use MQTT at my home as i got hass installed on pi.
But i would love still find some day any JSON for the 1L so i can use it in my garage or my GFs apartment.

mrmaximas, 2021-02-04:
I have flashed in to 1L https://github.com/mongoose-os-apps/shelly-homekit 
it's works!