RavenSystem/esp-homekit-devices #1239: List/report of devices wih HAA versions

wrx5, 2020-11-30:
Hi All,

I have 27 light switches around the house with HAA OTA firmware. Is there way to see their firmware version without put it into setup mode and connect one by one?

tonysprenk, 2020-11-30:
You can see it in settings in the home app.

tonysprenk, 2020-11-30:
Long press the accessory and swipe down.

pilot1981, 2020-11-30:
You can see both from Apple home app? Firmware HAA version and HAAT OTA version?

wrx5, 2020-11-30:
Hi Guys,

thanks for the responses. What I am after is not one by one approach. I was planning to write a simple HTML page where it accesses each device with a URL to get a response. 

For example: 

http://192.168.1.10/status.html?name
http://192.168.1.10/status.html?version

I can do this for all devices on one page to see all together instead of doing it one by one.

Thanks

closed 2020-12-30 by stale[bot]

wrx5, 2021-01-11:
> Hi Guys,
>
> thanks for the responses. What I am after is not one by one approach. I was planning to write a simple HTML page where it accesses each device with a URL to get a response.
>
> For example: [...]

No Luck with this?