RavenSystem/esp-homekit-devices #1242: Controlling HAA device from another HAA device

wrx5, 2020-12-02:
Hi All,

All my light switches are HAA. I needed a solution for the light at upstairs which has 2 controls. One at downstairs at the beginning of the stairs and the other one is at upstairs. I think this is called 2 way controlled light or something like that.

So what I did is I installed the main one upstairs which has direct cabling to the light. Works perfect. I installed another HAA switch downstairs but it doesn't directly controls the light. I can turn it on or off from my phone but nothing connected to load. Instead I created a 4 Home Automations:

1) If downstairs switch is on then Turn on Upstairs Switch (no time or people filter)
2) If downstairs switch is off then Turn off Upstairs Switch (no time or people filter)
3) If upstairs switch is on then Turn on Downstairs Switch (no time or people filter)
4) If upstairs switch is off then Turn off Downstairs Switch (no time or people filter)

The problem is this setup doesn't always work. Intermittently works. I don't know if there is any delay or waiting period between the actions but regardless of from up or down stairs, it doesn't work all the times.

Home app gets the status change from the switch that I press within split second. I can see it in the Home app. But automation doesn't kick in everytime. It just works whenever it feels like :) 

On the other hand, I have smart door sensors (Aqara) on certain doors to see the status of the door, Open close one (non-HAA). Home app gets its status change within split second. Very fast. Those door sensors also connected to related HAA light switches via automations in Home App. Automations are very simple:

If door is open then turn the related light on
If door is close then turn the related light off

Same, doesn't always work. 

Any thoughts?

H-Koenig-Deutschland, 2020-12-10:
you should be able to control the light from both switches in a two-way switch installation. The downstairs HAA is not necessary.
How did you wired the switches?

mwouda, 2021-01-07:
Hi , 

Do yo have a AppelTV 4 or ipad @home that function as a bridge? 
If you don't check to see if you phone is connected to you Wifi.. If not then that might be the problem.

I have a 2 way switch ([Sonoff wall switch 2 Way](https://www.itead.cc/smart-home/sonoff-tx-series-wifi-smart-wall-switches.html)) in my bedroom. Only one is connected. the other button is a Home Automation: 
If button is enable then activate scene: GoodNight. enable accessoires Bedroom.. Disable after 1 minute. 

This works like a charm.

wrx5, 2021-01-11:
> 
> 
> you should be able to control the light from both switches in a two-way switch installation. The downstairs HAA is not necessary.
> How did you wired the switches?

Upstairs is normal (one-way) wired as if only one switch.

Downstairs is source power and neutral is connected to power up the switch but nothing connected to L1. So I control its state in Home Automation. If anyone presses the switch automation turns the upstairs switch on. Basically I am using the downstairs switch as dummy switch.

I thought we cannot use two-way wiring with these devices as there is no common port.

wrx5, 2021-01-11:
> Hi ,
>
> Do yo have a AppelTV 4 or ipad @home that function as a bridge?
> If you don't check to see if you phone is connected to you Wifi.. If not then that might be the problem.
>
> I have a 2 way switch [...]

I did a search on internet and kinda found that iPad should be a last resort to use as HUB. we should use Apple TV or HomePod as hub. Found few people having similar issues. Not sure why iPad is having problem with automations. (if it is the problem)

closed 2021-02-10 by stale[bot]