RavenSystem/esp-homekit-devices #1243: Difference between HAA and HAA OTA releases?

wrx5, 2020-12-02:
Hi Everyone,

Forgive me for this "dummies" question but I wonder What's the difference between HAA and HAA OTA releases?

Today the latest version of HAA is 3.8.4 and HAA OTA is 4.3.0. 

When I put the device into setup mode and tick the Update All Now and Auto OTA Updates options then hit "save" button it only gets HAA releases not HAA OTA.

Thanks in advance

rbrtsluk, 2020-12-03:
no one knows?

tonysprenk, 2020-12-03:
HAA OTA is the part of the software that manages the update and install mechanism. There always needs to be two parts (HAA OTA and HAA) because when you update, HAA is overwritten with the new version. But if there is no HAA OTA there is no software to tell HAA where and how to be installed anymore. This is the simplest way I can think of to explain it, hopefully it makes sense.

wrx5, 2020-12-03:
So where do we use HAA OTA? can we flash the device with that one instead of HAA?

tonysprenk, 2020-12-04:
It’s part of fullhaaboot.bin so if you flash with that it’s included. The other option is to flash without OTA (only in special cases). More info here: https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation

xrust83, 2020-12-04:
https://www.samsung.com/my/support/mobile-devices/what-are-the-requirements-when-upgrade-firmware-with-fota-firmware-over-the-air/

ss83, 2020-12-26:
I used the HomeKitFlasher for two magichome devices. The first one I used the "Flash HAA v.4.XXX" button. The second one I used the "Flash OTA HAA Firmware".

Am I right that only the second one where I used the OTA button will be able to update through "AUTO OTA Updates" ? Is there a way to update the first one to the OTA version using the interface, e.g. under the custom server ??? I have already installed it into the wall, so it would be nice not to dismantle it again :-)

closed 2021-01-25 by stale[bot]