RavenSystem/esp-homekit-devices #1246: Use ADC (pin A0) as input

danielhelmstedt, 2020-12-08:
Bump of #542 and #1084.

Since 8266s have a single analog to digital converter on A0, It would be great to be able to filter and use the result of analogRead(A0) as  a stateless button input. Theoretically this would allow you to run a wire from the adc pin of any 8266 device to any metal surface and use it as a touch sensitive input to a stateless button.

danielhelmstedt, 2021-01-05:
Bumping this. Any thoughts? I've recently implemented an analogRead(A0) into one of my projects, that reads the analog output of a pressure sensitive mat, compares to a threshold, and sets an occupancy sensor accordingly. The actual read function is quite simple, I imagine it'd be relatively simple for you to implement a threshold as a JSON characteristic?

My snippet to read Analog value and return occupancy is here, for reference.
`void my_homekit_report() {
  int sensorValue;
  sensorValue = analogRead(A0);
  Serial.print("sensor = ");
  Serial.print(sensorValue);
  uint8_t occupancy = sensorValue < 100 ? 0 : 1; // The trigger logic. TODO: add threshold
  cha_occupancy.value.bool_value = occupancy;
  homekit_characteristic_notify(&cha_occupancy, cha_occupancy.value);
  LOG_D("occupancy %u", occupancy);
}`

RavenSystem, 2021-01-25:
Sorry, button library does not support it, and it will not do. Use other GPIO instead.

closed 2021-01-25 by RavenSystem