RavenSystem/esp-homekit-devices #1248: connection problem for leak detector which are activated at start-up

clint80, 2020-12-09:
Hello
I have configured leak detectors with an ESP8266.When I turn it on with no leak detected, everything works fine. I test my sensors and it indicates leaks in the homekit. if I put one or more sensors in water and that turns on nothing happens, impossible for the ESP8266 to connect and therefore impossible to switch to configuration mode with the integrated control. I have to put all the sensors out of the water and reconnect the ESP so that it works again.
is there a function which would take into account the detectors x time after starting the ESP
here is my JSON which works very well when the detectors are not activated at startup. {"n": "detecteur_fuite_comble_facade",
"a": [{"t": 7, "f1": [{"g": 1, "t": 0}], "f0": [{"g": 1}]}, {"t ": 7," f1 ": [{" g ": 2," t ": 0}]," f0 ": [{" g ": 2}]}, {" t ": 7," f1 ": [{"g": 3, "t": 0}], "f0": [{"g": 3}]}, {"i": 2, "1": {"s": [{" a ": 1}]}},
{"i": 2, "1": {"s": [{"a": 2}]}}]}

hillaliy, 2020-12-11:
Hi, add "xa":0
like this "t":7,"xa":0

clint80, 2020-12-11:
Hillaliy good evening and big thank you
all are working properly
wishing you happy end of year holidays and especially good health

closed 2020-12-11 by clint80