RavenSystem/esp-homekit-devices #1249: Garage door GPIO stays high after update

caiotadashi, 2020-12-09:
Hey guys

I was using the following JSON with my garage door and it was working fine.
Now i've updated to the latest version (v3.8.7) and my GPIO stays and keeps HIGH during and after boot.
Unfortunately what release it was before.
I've already flashed my NodeMCU again and the result is the same.

`{
  "c": {
    "ttl": 60,
    "l": 2,
    "b": [
      {
        "g": 0,
        "t": 5
      }
    ]
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 5,
            "v": 1,
            "i": 0.5
          }
        ]
      },
      "1": {
        "a": 0
      },
      "2": {
        "a": 0
      },
      "3": {
        "a": 0
      },
      "5": {
        "h": [
          {
            "h": "192.168.1.17",
            "p": 34567,
            "m": 2,
            "u": "haa/sendMessage?text=⚠️%20Port%C3%A3o%20fechado"
          }
        ]
      },
      "6": {
        "h": [
          {
            "h": "192.168.1.17",
            "p": 34567,
            "m": 2,
            "u": "haa/sendMessage?text=⚠️%20Port%C3%A3o%20aberto"
          }
        ]
      },
      "t": 40,
      "d": 7,
      "f3": [
        {
          "g": 12,
          "t": 0
        }
      ],
      "f4": [
        {
          "g": 12
        }
      ],
      "f2": [
        {
          "g": 14,
          "t": 0
        }
      ],
      "f5": [
        {
          "g": 14
        }
      ]
    }
  ]
}`

Aureliusz1, 2020-12-09:
Hi,
 I'm connecting to the topic. There is some bug in the software. After a power failure, GPIO12 goes high and opens the gate. It worked fine on version 3.4.1.

Joniss77, 2020-12-09:
This fixed it for me:

https://github.com/RavenSystem/esp-homekit-devices/issues/1249

RavenSystem, 2020-12-09:
Duplicated:

closed 2020-12-09 by RavenSystem