RavenSystem/esp-homekit-devices #1255: MCP23017 (SMD sized, PCB version, I2C problem)

peking84, 2020-12-14:
Hi @RavenSystem

Playing with MCP23017. It is not working for me.

Tried with different gpios, (5-4, 14-13, 2-0), different nodeMCU (LoLin / AmiCa) with the same SMD sized PCB MCP23017 (named from different brands, but the same board model WCMCU-2317; CJMCU-2317, ... , basically the expander chip with pullup resistors and capacitor. Also tried and removed those SMD elements and using with 3.3V only). Not to mention no success.

Connected pins:
GND-A0-A1-A2 -> GND (0x20)
                   SCL -> GPIO 5
                  SDA -> GPIO 4
            VCC,RST -> 3.3V or 5.0V

Using devtype 1 or 2, changing v=1/0 with g=100/101/102/103/108..., visible in the logs, but no change on the output.
Also changed 100kHz to 400kHz, no change.

I switched temorarily to a simple code to make it blink (ref: http://www.esp8266learning.com/esp8266-mcp23017-example.php).
And with this the extension "board" with the MCP23017 works.
Is there anyone out there who faced such issue? 

I found one ticket reporting about of a similar issue, but only with extended GPIOs B0->B7.

https://github.com/RavenSystem/esp-homekit-devices/issues/1215
You closed it 6 days ago with "There was a bug with channel B. It is fixed with last version v3.8.6."

I seem to have issue with all 16 port, even on the latest HAA v4.1.0. 
Is there any way to debug this further?

Example/Simple json, in the log below, which i have gone through several times but looks decent based on the json declaration.

```
JSON:
 {"c":{"o":4,"ic":[{"c":5,"d":4,"q":100}],"mc":[{"b":0,"a":32,"ca":255,"cb":255}]},"a":[{"1":{"r":[{"g":100,"v":1}]},"0":{"r":[{"g":100,"v":0}]}},{"1":{"r":[{"g":101,"v":1}]},"0":{"r":[{"g":101,"v":0}]}},{"t":2,"i":1,"0":{},"1":{"s":[{"a":1}],"b":[{"g":0}]}}]}

Hostname: HAA-XXXXXX
I2C bus: 0, scl: 5, sda: 4, freq: 100000, res: 0
MCP23017 0: bus: 0, addr: 32, chA: 255, chB: 255

:

[5] Update Characteristics
<2> Setter ON 1
<2> Exec Action 1
<2> Digital Output: gpio 101, val 1, inch 0
[5] Update Characteristics
<1> Setter ON 1
<1> Exec Action 1
<1> Digital Output: gpio 100, val 1, inch 0
```

The other thing: for this output device, there is no wiki at all. Will there be ?
(https://github.com/RavenSystem/esp-homekit-devices/releases/tag/HAA_3.6.0   has the explanation and Jose's new iDevice)

(Also looking forward for the support of other I2C devices !)

/Starmarks

Tomfresse, 2020-12-22:
Do you have pull-up resistors in I2C line ?
If not ... add two  ~2.7 kOhm resistors between SDA, SCL and  3.3V VCC.

peking84, 2020-12-24:
Yes. Already tried with other than the one preinstalled on the board which if i recall 10k’ish ones. Also tried with 3.1k and without pullup resistors. Ill give a shot with another ic after the holidays. Thanks for the hint anyway. Cheers

closed 2021-01-24 by stale[bot]