RavenSystem/esp-homekit-devices #1258: TYWE3S Tuya ESP8266 Water Valve Json

asbesto01, 2020-12-14:
Hello everybody, 

I would like to flash this water valve:

https://www.amazon.it/gp/product/B084VPD5TM/ref=ox_sc_saved_title_1?smid=A27C3VFUSTVDBL&psc=1

I only need water valve json to perform it; can someone point me in the right direction?

Thank you so much!

mrmaximas, 2020-12-21:
try to my config from similar valve, but pay attention for GPIO.
my valve with this GPIO 
GPIO 12 BUTTON
GPIO 13 RELAY
GPIO 0 LED 
{"c":{"o":0},"a":[{"0":{"r":[{"g":13,"v":0},{"g":0,"v":0}]},"1":{"r":[{"g":13,"v":1},{"g":0,"v":1}]},"t":20,"b":[{"g":12,"t":1,"p":1}],"s":5,"w":0,"d":0}]}

asbesto01, 2020-12-22:
@mrmaximas thank you so much!

Can you send me the link to your valve? So I can use your json purchasing an identical valve and I am sure I do not have problems :)

Thanks

mrmaximas, 2020-12-22:
i have 3 valves from 3 another sellers ))
https://www.aliexpress.com/item/10000055519067.html
https://www.aliexpress.com/item/4000902933958.html
https://www.aliexpress.com/item/4000170119421.html
first of them I had to flash into tasmota using wires, another two flashed via tuya convert

asbesto01, 2020-12-24:
> i have 3 valves from 3 another sellers ))
> 
> https://www.aliexpress.com/item/10000055519067.html
> 
> https://www.aliexpress.com/item/4000902933958.html
> 
> https://www.aliexpress.com/item/4000170119421.html
> 
> first of them I had to flash into tasmota using wires, another two flashed via tuya convert

After this tasmota and tuya convert steps have your been able to flash haa ravencore? That's what I want to achieve!

mrmaximas, 2020-12-24:
first - flash tasmota via tuya convert then flash ravencore haa 
BUT tuya-convert might not be possible for devices if the manufacturer started using newest tuya firmware.
this is possible only via wires soldering

asbesto01, 2020-12-24:
> first - flash tasmota via tuya convert then flash ravencore haa
> BUT tuya-convert might not be possible for devices if the manufacturer started using newest tuya firmware.
> this is possible only via wires soldering

how to check if the device has old or new tuya firmware? which version should I check?

Thanks

closed 2021-01-23 by stale[bot]

mrmaximas, 2024-04-22:
@asbesto01 may be you have original fw backup??

asbesto01, 2024-04-22:
Hi, I never flashed the valves, so the 5 valves I own are still with original tuta firmware, then I exposed them to HomeKit using a zemismart hub (zigbee).

everything is working perfectly!