RavenSystem/esp-homekit-devices #1259: TreatLife Dimmer DS02S

guatedude2, 2020-12-16:
Hey all, 
I've been looking for a way to configure the [TreatLife Dimmer DS02S](https://www.amazon.com/dp/B07PJTLB7Z/ref=cm_sw_em_r_mt_dp_inz2FbD2ZG6QZ?_encoding=UTF8&psc=1) with HAA firmware. I got it working with Tasmota using the TuyaMCU configuration. I'm not sure how can I send [TuyaMCU commands](https://tasmota.github.io/docs/TuyaMCU/#data-type-table) to the device using the accessory configuration. It would be great to get these working without HomeBridge.

Here's the config I use in Tasmota:
[image]

tonysprenk, 2020-12-16:
You will need to know the raw commands and configure your json according to the serial commands section in the wiki. However: HAA only supports sending commands, not receiving like tasmota. What this means is that if this device has a physical button, when you press it the secondary mcu will send a command to the esp chip but HAA can’t process this so the status won’t be updated in the home app. Example: you have the dimmer configured in the home app and set it to 50%, the lamp goes to 50%. Then you use the physical button to set it to 100%, the lamp goes to 100% but the home app stays at 50%. 

Receiving serial commands is not supported by HAA because there are many different protocols. The argument has been put forward to at least support the tuya protocol as this would open up HAA to many more device like sensors and dimmers etc. @RavenSystem has stated that if someone can implement the protocol in a clean way he will accept the fork. So if you or someone you know is willing to code this head over to #915 for more info. Hope this helps answer your question. If so please close this issue. Thanks!

guatedude2, 2020-12-16:
That's great. I would love to help contribute but my Arduino knowledge is very limited since I specialize in frontend. I can give it a try but it would probably take me twice as long as someone with more domain knowledge. Maybe something I can coordinate with someone with more knowledge.

guatedude2, 2020-12-16:
Looks like Tasmota's TuyaMCU driver is written in C 
https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_16_tuyamcu.ino this could be promising to enable a way to configure the TuyaMCU.

tonysprenk, 2020-12-17:
That is promising. Would you mind posting your findings in #915 ?

guatedude2, 2020-12-17:
Sure thing

paullj1, 2021-01-02:
Might be the Tuya MCU protocol definition:
https://images.tuyacn.com/smart/aircondition/Guide-to-Interworking-with-the-Tuya-MCU.pdf

closed 2021-02-18 by stale[bot]

paullj1, 2021-03-06:
https://github.com/paullj1/TreatLife-DS02S-HomeKit/tree/main

mkormendy, 2024-06-28:
> https://github.com/paullj1/TreatLife-DS02S-HomeKit/tree/main

But now you have to open it up and flash with wires. Can't flash OTA if you installed HAA firmware because of code signing check on OTA process. *dumb in my opinion*