RavenSystem/esp-homekit-devices #1266: Garage door default config issue with closing state

mmdumi, 2020-12-20:
Hi,
I'm trying to use the default config (from wiki) for garage door setup on a Sonoff SV device.
What happens is that the device switches correctly from `opening` to `open` state but when it's time for closing it remains indefinitely in the `closing` state. 
Can you please help configure it correctly to switch to closed state ?
```
{
  "c":{"l":13,"b":[{"g":0,"t":5}],"0":{"r":[{"g":12}]}},
  "a":[{
    "t":40,
    "d":18,
    "0":{"r":[{"g":12,"v":1,"i":0.5}]},
    "1":{"a":0},
    "2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},
    "3":{"a":2},
    "f3":[{"g":14,"t":0}],
    "f4":[{"g":14,"t":1}]
  }]
}
```

samialdury, 2020-12-22:
Also having this issue😕

samialdury, 2020-12-22:
@mmdumi Found a quick solution. 
You just need to remove the State inputs (`"f3"` & `"f4"` in this example) and then it works as expected. Correct example as follows:
```json
{
  "c":{"l":13,"b":[{"g":0,"t":5}],"0":{"r":[{"g":12}]}},
  "a":[{
    "t":40,
    "d":18,
    "0":{"r":[{"g":12,"v":1,"i":0.5}]},
    "1":{"a":0},
    "2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},
    "3":{"a":2}
  }]
}
````

mmdumi, 2020-12-23:
@samialdury Thank you for your response. Indeed it works correctly with the above configuration.
What I would like to figure out next is how to change the state of the sonoff device when the door is controlled with the regular rf transmiter. 
Example: if I open the garage door with homekit sonoff will be marked as `opened` but if I use the rf transmitter to close it sometime later, the door will still say open. :( 
If I can't get this to work correctly somehow I would just need to reflash the sonoff device with a regular switch configuration.

morphix9, 2020-12-26:
If you remove the state inputs (sensors), then there is no way Homekit could know you have opened/closed the door with a RF or somehow else.

closed 2021-01-25 by stale[bot]