RavenSystem/esp-homekit-devices #1267: Problem Configuration

TheReiner, 2020-12-20:
Hi,

Firs, english is not my native language...
i have this problem, i have a bulb CW-WW, but my configuration don't work. After save, the bulb reboot on setup mode.

{“c":{"q":850,"b":[{"g":0,"t":5}]},"a":[{"t":30,"cw":14,"ww":12,"b":[{"g":0}]},{"t":1,"1":{"s":[{"a":1}]}}]}

Thanks a lot :)

RavenSystem, 2020-12-20:
First quote has a bad format.

TheReiner, 2020-12-20:
Ok... It's better ^^
No reboot on setup mode, but my bulb don't work. Nothing ...

If i use this configuration like RGB
{"c":{"q":850,"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":14,"g":12,"v":5,"b":[{"g":0}]},{"t":1,"1":{"s":[{"a":1}]}}]}
It's work .. But with bad colour (it's Logic ^^)

How to do that ? is it possible like this ?
[image]

TheReiner, 2021-01-15:
Hi,
No Solution for me ?
Thanks :)

Kienz, 2021-01-15:
Some people are working on better colors - there is also a beta HAA version. You can join the Discord chat https://discord.gg/v8hyxj2 (Channel "lightning").
In the latest beta 5.50.13 is a bug - so don't use this and wait until next beta is released.

closed 2021-01-25 by RavenSystem