RavenSystem/esp-homekit-devices #1269: Problem with OTA Update

mrmaximas, 2020-12-21:
hi!
i have a problem with OTA Update of Water Valve with this config
{"c":{"o":0,"b":[{"g":12,"t":5}],"m":20},"a":[{"0":{"r":[{"g":13,"v":0},{"g":0,"v":0}]},"1":{"r":[{"g":13,"v":1},{"g":0,"v":1}]},"t":20,"b":[{"g":12,"t":1,"p":1}],"s":5,"w":0,"d":0}]}
One of three valves not able to update to latest HAA 4.3.3
I put the valve in configuration mode, check the "Update All Now" box and click the "Save" button, but no update occurs after that.  I wait a long time, but the old version of HAA 3.0.0 remains
What i doing wrong?
Thanks.
[image]

mrmaximas, 2020-12-22:
i'm truing to ota update via local web server, but nothing to update after download
here is log file 
192.168.88.163 - - [22/Dec/2020 16:00:55] "GET /otaversion HTTP/1.1" 200 -
192.168.88.163 - - [22/Dec/2020 16:00:57] "GET /haaboot.bin.sec HTTP/1.1" 200 -
192.168.88.163 - - [22/Dec/2020 16:01:00] "GET /haaversion HTTP/1.1" 200 -
192.168.88.163 - - [22/Dec/2020 16:01:03] "GET /haamain.bin.sec HTTP/1.1" 200 -

mrmaximas, 2020-12-22:
here is log from github update
0003.414 192.168.88.163-055F4C: DNS check result  = OK
0005.415 192.168.88.163-055F4C:
0005.416 192.168.88.163-055F4C: *** STARTING UPDATE PROCESS
0005.417 192.168.88.163-055F4C:
0005.417 192.168.88.163-055F4C:
0005.418 192.168.88.163-055F4C: Running OTAMAIN
0005.431 192.168.88.163-055F4C: Host: github.com
0005.433 192.168.88.163-055F4C: Range: bytes=0-1
0009.423 192.168.88.163-055F4C: NEW CONNECTION LocalPort=f4b1 DNS..OK IP addr: 140.82.121.3 Local..OK Remote..OK SSL..OK set_fd to github.com port 443..OK
0009.426 192.168.88.163-055F4C: sent OK
0009.521 192.168.88.163-055F4C: ret = 1024
0009.522 192.168.88.163-055F4C:
0009.523 192.168.88.163-055F4C: HTTP/1.1 302 Found
0009.524 192.168.88.163-055F4C: date: Tue, 22 Dec 2020 15:00:28 GMT
0009.525 192.168.88.163-055F4C: content-type: text/html; charset=utf-8
0009.558 192.168.88.163-055F4C: expect-ct: max-age=2592000, report-uri="https://api.github.com/_private/browser/errors"
.githubstatus.com collector.githubapp.com api.github.com github-cloud.s3.amazonaws.com github-production-repository-file-5c1aeb.s3.amazonaws.com github-production-upload-manifest-file-7fdce7.s3.amazonaws.com github-production-user-asset-6210df.s3.amazonaws.com cdn.optimizely.com logx
0009.602 192.168.88.163-055F4C:
0009.603 192.168.88.163-055F4C: HTTP returns 302
0009.611 192.168.88.163-055F4C: GET /RavenSystem/haa/releases/download/4.3.4/otaversion HTTP/1.1
0013.496 192.168.88.163-055F4C: NEW CONNECTION LocalPort=f4b2 DNS..OK IP addr: 140.82.121.3 Local..OK Remote..OK SSL..OK set_fd to github.com port 443..OK
0013.499 192.168.88.163-055F4C: sent OK
0013.638 192.168.88.163-055F4C: ret = 1024
0013.639 192.168.88.163-055F4C:
0013.640 192.168.88.163-055F4C: HTTP/1.1 302 Found
0013.641 192.168.88.163-055F4C: date: Tue, 22 Dec 2020 15:00:30 GMT
0013.641 192.168.88.163-055F4C: content-type: text/html; charset=utf-8
0013.642 192.168.88.163-055F4C: server: GitHub.com
0013.643 192.168.88.163-055F4C: status: 302 Found
0013.723 192.168.88.163-055F4C: Forwarding: github-production-release-asset-2e65be.s3.amazonaws.com/166993980/9bb75f80-445f-11eb-86b9-1735668bafa7?X-Amz-Algorit0014.744 192.168.88.163-055F4C: NEW CONNECTION LocalPort=f4b3 DNS..OK IP addr: 52.217.85.68 Local..OK Remote..OK SSL..OK set_fd to github-production-release-asset-2e65be.s3.amazonaws.com port 443..OK
0019.815 192.168.88.163-055F4C: Content-Disposition: attachment; filename=otaversion
029Z&X-Amz-Expires=300&X-Amz-Signature=a21925e13b63397a376a4f430c5e4cfeec7575f93bc3a7ff98d0ebcff100e675&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dotaversion&response-content-type=application%2Foctet-stream
0020.467 192.168.88.163-055F4C: NEW CONNECTION LocalPort=f4b4 DNS..OK IP addr: 52.217.85.68 Local..OK Remote..OK SSL..OK set_fd to github-production-release-asset-2e65be.s3.amazonaws.com port 443..OK
0025.146 192.168.88.163-055F4C:  Downloaded 5 Bytes
0025.147 192.168.88.163-055F4C:
0025.149 192.168.88.163-055F4C: VERSION of otaversion: 4.3.1
0029.049 192.168.88.163-055F4C: NEW CONNECTION LocalPort=f4b5 DNS..OK IP addr: 140.82.121.3 Local..OK Remote..OK SSL..OK set_fd to github.com port 443..OK
0029.054 192.168.88.163-055F4C: sent OK
0029.264 192.168.88.163-055F4C: ret = 1024
0029.265 192.168.88.163-055F4C:
0029.266 192.168.88.163-055F4C: HTTP/1.1 302 Found
0029.267 192.168.88.163-055F4C: date: Tue, 22 Dec 2020 15:02:11 GMT
0029.267 192.168.88.163-055F4C: content-type: text/html; charset=utf-8
0029.291 192.168.88.163-055F4C: x-content-type-options: nosniff
0029.294 192.168.88.163-055F4C: x-xss-protection: 1; mode=block
0029.309 192.168.88.163-055F4C: content-security-policy: default-src 'none'; base-uri 'self'; block-all-mixed-content; connect-src 'self' uploads.github.com www.githubstatus.com collector.githubapp.com api.github.com github-cloud.s3.amazonaws.com github-production-repository-file-5c1aeb.s3.amazonaws.com github-production-upload-manifest-file-7fdce7.s3.amazonaws.com github-production-user-asset-6210df.s3.amazonaws.com cdn.optimizely.com
0029.344 192.168.88.163-055F4C:
0029.346 192.168.88.163-055F4C: HTTP returns 302
0029.347 192.168.88.163-055F4C:
0029.354 192.168.88.163-055F4C: GET /RavenSystem/haa/releases/download/4.3.4/haaboot.bin.sec HTTP/1.1
0029.360 192.168.88.163-055F4C: Host: github.com
0029.362 192.168.88.163-055F4C: Range: bytes=0-1
0033.536 192.168.88.163-055F4C: NEW CONNECTION LocalPort=f4b6 DNS..OK IP addr: 140.82.121.3 Local..OK Remote..OK SSL..OK set_fd to github.com port 443..OK
0033.825 192.168.88.163-055F4C: HTTP/1.1 302 Found
50215Z&X-Amz-Expires=300&X-Amz-Signature=ee1ee16b42aa18e246f0bb5151d593732a6f744ab85653812a818295d43348b2&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dhaaboot.bin.sec&response-conten=attachment%3B%20filename%3Dhaaboot.bin.sec&response-content-type=application%2Foctet-stream HTTP/1.1
0033.995 192.168.88.163-055F4C: Host: github-production-release-asset-2e65be.s3.amazonaws.com
0034.531 192.168.88.163-055F4C: NEW CONNECTION LocalPort=f4b7 DNS..OK IP addr: 52.217.48.116 Local..OK Remote..OK SSL..OK set_fd to github-production-release-as0039.030 192.168.88.163-055F4C: sent OK
0039.247 192.168.88.163-055F4C: ret = 471
0039.248 192.168.88.163-055F4C:
0039.249 192.168.88.163-055F4C: HTTP/1.1 206 Partial Content
0039.249 192.168.88.163-055F4C: x-amz-id-2: aSNhFhM3lbMHWjlVXOs4cPgmymsPmoXlUhGjWeCs0KnjSR0S6L40o3tTZMGHBohzVDZVJO5uDIo=
0039.269 192.168.88.163-055F4C: Content-Range: bytes 0-1/104
0039.328 192.168.88.163-055F4C:
0039.661 192.168.88.163-055F4C: NEW CONNECTION LocalPort=f4b8 DNS..OK IP addr: 52.217.48.116 Local..OK Remote..OK SSL..OK set_fd to github-production-release-asset-2e65be.s3.amazonaws.com port 443..OK
0044.335 192.168.88.163-055F4C:  Downloaded 104 Bytes

RavenSystem, 2020-12-22:
It is a bug that was fixed hours after HAA 3.0.0, but I see that you installed it. Solution? Try, and try, and try update. It will update.

mrmaximas, 2020-12-22:
> It is a bug that was fixed hours after HAA 3.0.0, but I see that you installed it. Solution? Try, and try, and try update. It will update.

I need to try this from github or possible from local web server?

RavenSystem, 2020-12-22:
It is the same. Bug is related to network logs that causes a system crash. Some users could update after a couple of hours trying.

mrmaximas, 2020-12-22:
> It is the same. Bug is related to network logs that causes a system crash. Some users could update after a couple of hours trying.

Do I understand correctly that an update will be requested each time after power cycle?

RavenSystem, 2020-12-22:
No. You must put your device in setup mode each time.

mrmaximas, 2020-12-22:
> No. You must put your device in setup mode each time.

ok. i'm using short power cycle (power off, then short power on for 1-1.5sec then power off and power on)
it's ok? my json does not work correctly with 8s button press ((
{"c":{"o":0,"b":[{"g":12,"t":5}],"m":20},"a":[{"0":{"r":[{"g":13,"v":0},{"g":0,"v":0}]},"1":{"r":[{"g":13,"v":1},{"g":0,"v":1}]},"t":20,"b":[{"g":12,"t":1,"p":1}],"s":5,"w":0,"d":0}]}
this is valve with 
GPIO 12 BUTTON
GPIO 13 RELAY
GPIO 0 LED

mrmaximas, 2020-12-22:
> No. You must put your device in setup mode each time.

it's good result in log from local server?
Advanced ESP Logger (c) 2020 José Antonio Jiménez Campos

0005.267 192.168.88.163-055F4C:
0005.270 192.168.88.163-055F4C:
0005.271 192.168.88.163-055F4C:
0005.271 192.168.88.163-055F4C: Running OTAMAIN
0005.272 192.168.88.163-055F4C:
0008.031 192.168.88.163-055F4C: NEW CONNECTION LocalPort=e17a DNS..OK IP addr: 192.168.88.231 Local..OK Remote..OK
0008.033 192.168.88.163-055F4C: sent OK
0008.050 192.168.88.163-055F4C: ret = 17
0008.053 192.168.88.163-055F4C:
0008.053 192.168.88.163-055F4C: HTTP/1.0 200 OK
0011.046 192.168.88.163-055F4C: NEW CONNECTION LocalPort=e17b DNS..OK IP addr: 192.168.88.231 Local..OK Remote..OK
0011.048 192.168.88.163-055F4C: sent OK
0011.084 192.168.88.163-055F4C: ret = 17
0011.085 192.168.88.163-055F4C:
0011.086 192.168.88.163-055F4C: HTTP/1.0 200 OK
0011.087 192.168.88.163-055F4C:
0014.047 192.168.88.163-055F4C: NEW CONNECTION LocalPort=e17c DNS..OK IP addr: 192.168.88.231 Local..OK Remote..OK
0014.049 192.168.88.163-055F4C: sent OK
0014.075 192.168.88.163-055F4C: ret = 17
0014.077 192.168.88.163-055F4C: HTTP/1.0 200 OK
0014.077 192.168.88.163-055F4C:
0014.080 192.168.88.163-055F4C: Get sign
0017.047 192.168.88.163-055F4C: NEW CONNECTION LocalPort=e17d DNS..OK IP addr: 192.168.88.231 Local..OK Remote..OK
0017.049 192.168.88.163-055F4C: sent OK
0017.088 192.168.88.163-055F4C: ret = 17
0017.089 192.168.88.163-055F4C:
0017.090 192.168.88.163-055F4C: HTTP/1.0 200 OK
0017.091 192.168.88.163-055F4C:
0017.097 192.168.88.163-055F4C: Restarting...

RavenSystem, 2020-12-22:
Log is right, but is telling that your server has not partial content feature, that is needed.

mrmaximas, 2020-12-22:
> Log is right, but is telling that your server has not partial content feature, that is needed.

we did it! thanks!
0144.990 192.168.88.163-055F4C: Verifying sign...
0158.417 192.168.88.163-055F4C: Sign result: OK (1)
0158.417 192.168.88.163-055F4C: Finalize file
0158.419 192.168.88.163-055F4C:
0158.420 192.168.88.163-055F4C: *** HAAMAIN v4.3.4 installed
0158.420 192.168.88.163-055F4C:
0158.421 192.168.88.163-055F4C:
0158.421 192.168.88.163-055F4C: Restarting...

stefanathome, 2021-01-06:
I don't know - but assume to have a similar problem. I flashed the latest version of fullhaaboot.bin onto a Chinese PowerPlug. Above it was told to wait... the update will happen... But - after the boot process it hangs in an update loop. After downloading the latest update the **"Verifying sign" fails** and the update process starts again...
Can somebody help?
[image]

The following log repeats again and again.... (sorry for that long list)
_after downloading the firmware......._
17:19:25.464 -> Verifying sign...
17:19:37.938 -> Sign result: ERROR (0)
17:19:37.976 -> 
17:19:37.976 -> !!! Error installing OTAMAIN
17:19:37.976 -> 
17:19:37.976 -> 
17:19:37.976 -> Restarting...
17:19:38.975 -> rm match
17:19:38.975 -> del if0
17:19:38.975 -> usl
17:19:38.975 -> sul 0 0
17:19:39.051 -> 
17:19:39.051 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,1)
17:19:39.051 -> 
17:19:39.051 -> load 0x40100000, len 2292, room 16 
17:19:39.051 -> tail 4
17:19:39.051 -> chksum 0x57
17:19:39.051 -> load 0x3ffe8000, len 772, room 4 
17:19:39.051 -> tail 0
17:19:39.051 -> chksum 0x0b
17:19:39.051 -> csum 0x0b
17:19:39.051 -> 
17:19:39.051 -> rBoot v1.4.0 - richardaburton@gmail.com
17:19:39.051 -> Flash Size:   8 Mbit
17:19:39.051 -> Flash Mode:   DOUT
17:19:39.051 -> Flash Speed:  40 MHz
17:19:39.051 -> rBoot Option: Big flash
17:19:39.051 -> rBoot Option: RTC data
17:19:39.088 -> 
17:19:39.088 -> Booting rom 0.
17:19:39.344 -> pp_task_hdl : 3ffefb30, prio:14, stack:512
17:19:39.344 -> pm_task_hdl : 3ffef420, prio:1, stack:176
17:19:39.344 -> frc2_timer_task_hdl:0x3fff4338, prio:12, stack:200
17:19:39.344 -> 
17:19:39.344 -> ESP-Open-SDK ver: 0.0.1 compiled @ Dec  8 2020 17:59:20
17:19:39.344 -> phy ver: 273, pp ver: 8.3
17:19:39.344 -> 
17:19:39.344 -> sleep disable
17:19:39.344 -> 
17:19:39.344 -> 
17:19:39.344 -> 
17:19:39.379 -> Sysparam OK
17:19:39.379 -> 
17:19:39.379 -> Wifi Init
17:19:39.379 -> mode : sta(b8:f0:09:01:e1:6b)
17:19:39.379 -> add if0
17:19:39.414 -> Saved BSSID: 7cff4d8aefbc
17:19:39.414 -> Wifi Mode: Normal
17:19:39.414 -> 
17:19:39.414 -> HAA OTA - SETUP MODE
17:19:39.414 -> 
17:19:39.452 -> mode : sta(b8:f0:09:01:e1:6b) + softAP(ba:f0:09:01:e1:6b)
17:19:39.452 -> add if1
17:19:39.452 -> bcn 100
17:19:39.452 -> Function called without core lock
17:19:39.452 -> Start AP SSID=HAA-01E16B
17:19:39.452 -> bcn 0
17:19:39.452 -> del if1
17:19:39.452 -> add if1
17:19:39.452 -> bcn 100
17:19:39.452 -> Start DHCP server
17:19:39.452 -> Start DNS server
17:19:39.452 -> Start HTTP server
17:19:41.044 -> scandone
[... 811 lines cut ...]
17:26:33.901 -> Sector 0xdd000  Downloaded 319488 Bytes
17:26:34.343 -> Sector 0xde000  Downloaded 323584 Bytes
17:26:34.787 -> Sector 0xdf000  Downloaded 327680 Bytes
17:26:35.158 -> Sector 0xe0000  Downloaded 331776 Bytes
17:26:35.562 -> Sector 0xe1000  Downloaded 335872 Bytes
17:26:36.046 -> Sector 0xe2000  Downloaded 339968 Bytes
17:26:36.486 -> Sector 0xe3000  Downloaded 344064 Bytes
17:26:36.892 -> Sector 0xe4000  Downloaded 348160 Bytes
17:26:37.299 -> Sector 0xe5000  Downloaded 352256 Bytes
17:26:37.814 -> Sector 0xe6000  Downloaded 356352 Bytes
17:26:38.223 -> Sector 0xe7000  Downloaded 360448 Bytes
17:26:38.630 -> Sector 0xe8000  Downloaded 364544 Bytes
17:26:39.036 -> Sector 0xe9000  Downloaded 368640 Bytes
17:26:39.450 -> Sector 0xea000  Downloaded 372736 Bytes
17:26:39.859 -> Sector 0xeb000  Downloaded 376832 Bytes
17:26:40.269 -> Sector 0xec000  Downloaded 380928 Bytes
17:26:40.681 -> Sector 0xed000  Downloaded 385024 Bytes
17:26:41.092 -> Sector 0xee000  Downloaded 389120 Bytes
17:26:41.534 -> Sector 0xef000  Downloaded 393216 Bytes
17:26:41.869 -> Sector 0xf0000  Downloaded 394784 Bytes
17:26:41.869 -> 
17:26:41.869 -> Verifying sign...
17:26:54.357 -> Sign result: ERROR (0)
17:26:54.357 -> 
17:26:54.357 -> !!! Error installing OTAMAIN

Kienz, 2021-01-15:
I had this issues on some devices too. Can't solve it - have to "reflash" the device.
https://github.com/RavenSystem/esp-homekit-devices/issues/825
Follow the steps 1-5 - after this it works - no re-pairing in HK is needed.

RavenSystem, 2021-01-25:
Some flash storage corruption.

@Kienz You are right.

closed 2021-01-25 by RavenSystem