RavenSystem/esp-homekit-devices #1270: Acce not found in home search

hanston, 2020-12-22:
have a light which had the unsupported wifi module it is converted to esp12f module
using this tutorial the light works great in tasmota 
converted to haa with json 
{ “c:” {“b”:[{“g”:0}]},”a”:[{“t”:30,“v”:13,“w”:09}]}
problem is this is not seen during search in home app says acce not found and times out
used this tutorial for sucessfull conversion
https://blakadder.com/frankenstein-saturday-3/

MystikalGinger, 2020-12-29:
Also having issues where I can flash HAA, set wifi information and JSON string but the device is never found in home search. I even went as far as to specify the accessory be visible to home even though it should be defaulted to this value. Not sure what im doing wrong here. 

{"c": {"b": [{"g": 0,"t": 5}]},"a": [{"h":1,"t": 30,"r": 5,”g”: 13,”v”: 12, ”b”: [{"g": 0}]}]}

Magichome RGB LED strip controller

MystikalGinger, 2021-01-02:
very frustrating that we cant get any input on this issue

hanston, 2021-01-03:
this is what i have observed
make sure the json string is correct
what i did is try the same from different devices and i was able to get it to work
IF device is not found MOST probably the string input is not proper 
i found this to b the reason

MystikalGinger, 2021-01-04:
Yes, I came to the conclusion that is most likely the reason as well. Trying to find some resources to help as after reviewing documentation I do not see any errors in the string I am using.

Kienz, 2021-01-15:
@MystikalGinger Your JSON is wrong - wrong double qoutes for the properties g, v and b used.
Try this JSON:
```json
{"c": {"b": [{"g": 0,"t": 5}]},"a": [{"h":1,"t": 30,"r": 5,"g": 13,"v": 12, "b": [{"g": 0}]}]}
```

closed 2021-02-14 by stale[bot]