RavenSystem/esp-homekit-devices #1271: Thingsboard.io + HAA post command

adamkasprzak, 2020-12-22:
Hi! 

I'm logging data from a few sensors to my Thingsboard server with Arduino and post commands. I have read [wiki](https://github.com/RavenSystem/esp-homekit-devices/wiki/Accessory-Configuration#Send-HTTP-Request-Actions) and it is technically possible to log data from HAA thermostats with CURL post command. There is an instruction on [Thingsboard.io](https://thingsboard.io/docs/reference/http-api/):

`curl -v -X POST -d @telemetry-data-as-object.json http://localhost:8080/api/v1/$ACCESS_TOKEN/telemetry --header "Content-Type:application/json"
`

What I'm missing is the ability to set header. My current post config looks like this:

        "y0": [{
            "v": 0,
            "r": 1,
            "0": { "h": [{ "h": "serv_ip_addr", "p": 8080, "m": 2, "u": "/api/v1/__token__/telemetry", "c": "{\"temp\":#HAA@0000,\"hum\":#HAA@0001}" }] }
        }]

I have also red this topic https://github.com/RavenSystem/esp-homekit-devices/issues/1029 which ended with having to use alternate solution (TCP RAW) - could you help with that? I was googling without any success.

Thanks in advance!

RavenSystem, 2020-12-24:
FYI: https://github.com/RavenSystem/esp-homekit-devices/releases/tag/HAA_4.4.0

closed 2021-01-23 by stale[bot]

adamkasprzak, 2021-02-19:
I have tried using the new function

`"e": "Content-type: application/json\r\n",`

but there is nothing received on the other side.
My complete json for Sonoff TH is:

`{
  "c": { "l": 13, "b": [{ "g": 0, "t": 5 }] },
  "a": [{
    "t": 1,
    "s": 5,
    "0": { "r": [{ "g": 12, "v": 0 }] },
    "1": { "r": [{ "g": 12, "v": 1 }] },
    "b": [{ "g": 0 }]
  },
    {
    "t": 24,
    "g": 14,
    "n": 4,
    "j": 30,
    "z": 0,
    "y0": [{
        "v": 0,
        "e": "Content-type: application/json\r\n",
        "r": 1,
        "0": { "h": [{ "h": "server.domain", "p": 8080, "m": 2, "u": "/api/v1/___TOKEN___/telemetry", "c": "{\"temp_test\":#HAA@0000,\"hum_pokoj\":#HAA@0001}" }] }
    }]
  }]
}`

working CURL:
`curl -v -X POST -d "{\"temp_test\": 1}" http://server.domain:8080/api/v1/___TOKEN___/telemetry --header "Content-Type:application/json"`

@RavenSystem - Am I using the "e" parameter right?

I can not see any logs, HAA v. is 5.0.3

adamkasprzak, 2021-02-19:
I have tried with dummy numbers instead of #HAA@000/1 and nothing changed. I guess the action is not triggered.

adamkasprzak, 2021-02-19:
Ok, finally got it working!! Below is working code for Sonoff TH with working POST with JSON header which can be used with wonderful [Thingsboard](https://thingsboard.io) system. Cheers!

{
  "c": { "l": 13, "b": [{ "g": 0, "t": 5 }] },
  "a": [{
    "t": 1,
    "s": 5,
    "0": { "r": [{ "g": 12, "v": 0 }] },
    "1": { "r": [{ "g": 12, "v": 1 }] },
    "b": [{ "g": 0 }]
  },
    {
    "t": 24,
    "g": 14,
    "n": 4,
    "j": 30,
    "z": 0,
    "y0": [{
        "v": 0,
        "r": 1,
        "0": {
            "h": [{
                "h": "YOUR_SERVER_IP_OR_DOMAIN",
                "p": 8080,
                "m": 2,
                "e": "Content-type: application/json\r\n",
                "u": "/api/v1/___YOUR_TOKEN_FROM_THINGSBOARD___/telemetry",
                "c": "{\"temp\":#HAA@0000,\"hum\":#HAA@0001}"
            }]
        }
    }]
  }]
}