RavenSystem/esp-homekit-devices #1275: Window Covering with Shelly 2.5 triggering relay on reboot

morphix9, 2020-12-29:
Hey guys,

I'm experimenting with the Window covering and a Shelly 2.5.

I'm using this HAA config (basically from the wiki example):

`{
  "c": {
    "0": {
      "r": [
        {
          "g": 16
        }
      ]
    },
    "l": 0,
    "b": [
      {
        "g": 2,
        "p": 0,
        "t": 5
      }
    ]
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 15,
            "v": 1
          },
          {
            "g": 4
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 15
          },
          {
            "g": 4,
            "v": 1
          }
        ]
      },
      "2": {
        "r": [
          {
            "g": 15
          },
          {
            "g": 4
          }
        ]
      },
      "3": {
        "r": [
          {
            "g": 15,
            "i": 0.2
          },
          {
            "g": 4
          }
        ]
      },
      "4": {
        "r": [
          {
            "g": 15
          },
          {
            "g": 4,
            "i": 0.2
          }
        ]
      },
      "t": 45,
      "o": 18,
      "c": 14,
      "f": 70,
      "xa": 0,
      "f0": [
        {
          "g": 5,
          "i": 1,
          "p": 0,
          "t": 0
        }
      ],
      "f1": [
        {
          "g": 13,
          "i": 1,
          "p": 0,
          "t": 0
        }
      ],
      "f2": [
        {
          "g": 5
        },
        {
          "g": 13
        }
      ]
    }
  ]
}`

As you see I'm using the `"xa": 0` parameter, but the opening relay is getting triggered every time I reboot the device. It's turned on, until a action is triggered. This means the window covering is opening on every restart (electricity outage for example). 

Do you have an idea what's wrong?

WojtekWro, 2020-12-29:
Hi, 
I’ve just realised that all my sonoff devices with more than one relay do this same. 

I’m pretty sure that behaviour is quite new. I’m wondering if this is a feature or a bug. 

@RavenSystem what do you think about it?

RavenSystem, 2020-12-30:
Feature. I updated device-database but not Wiki example. Now I have updated Wiki too.

WojtekWro, 2020-12-30:
> Feature. I updated device-database but not Wiki example. Now I have updated Wiki too.

@RavenSystem could you point out the exact place where it’s explained?
I could not find it in the Wiki. 
To be exact, my sonoff 4ch after connecting to the electricity send state high to a GPIO 15 (relay 4) for a second. I’m wondering why? 

In my case I’m going to use this relay to the garage door. But with this feature it is unacceptable. It will open the door every time after lack of electricity.

morphix9, 2020-12-30:
@RavenSystem is this related to my issue: https://github.com/RavenSystem/esp-homekit-devices/issues/1253

morphix9, 2020-12-30:
> @RavenSystem could you point out the exact place where it’s explained?
> I could not find it in the Wiki.
> To be exact, my sonoff 4ch after connecting to the electricity send state high to a GPIO 15 [...]

Check out the revisions. The window covering example has been changed:
[image]

But not documented what this new `"cd": 2` argument is meaning.

WojtekWro, 2020-12-30:
> But not documented what this new `"cd": 2` argument is meaning.

This is a 2 sec delay in starting accessory. But it doesn't change anything.

shanghaix, 2021-01-06:
I have the same issue, updated to the latest version, using shelly 2.5 for my blinds. Now when the electricity restarts the blinds move for a second or less meaning some power goes through.

closed 2021-01-14 by morphix9