RavenSystem/esp-homekit-devices #1277: Strange behavior on Siri / scenes

ss83, 2020-12-30:
Hi,

until now I have flashed 4 devices using HAA (4.4.1). 2 Gosund sockets and 2 magichome LED controllers.
After a bumpy start the two Gsound sockets now seem to work properly (first they got sometimes "no respond").

With the two magichome controllers I am still messing up. First of all they have often "no respond", but I cant figuere out if it is related to a specific state or something I do. I set already "ttl":60 (running a complete unifi Network, mDns is activated and other homekit devices work like a charm).
The second point I notices with the magichome devices is that they can only be controlled directly in the Home App. Neither a command by Siri (on the phone or on the Homepod) can drive them nor are the included in scenes. I just get the answer "one device is not available / does not react".

Any clue what can be checked? How to track down the issue? Do you think it is realted to HAA, to the magichome modules or evend directly to Homekit itself?

Regars Sven

DFC1990, 2020-12-30:
Hello what do you do with the "no respond" from godsund? because I got many times no respond

ss83, 2020-12-31:
Nothing special. After some time this solved by itself. Just the magichome controllers keep on strange behavior. Especially that they won't react in siri commands, but can be changed over the home app is not logical for me.

duyuelryel, 2021-01-02:
Greetings! I had the same issue too. After 2 days of start and item add to the homekit on iphone, the devices not responds. I use 3 and all do the same. If i enter to setup mode and hit save, then i can re-add them to homekit and they work for 2 days, then stop.
If i push the button on th-10, then i can see the light bulb lights, but the th-10 not responds, still. Strange.
Any ideas? Thank you!

DFC1990, 2021-01-02:
Hi all, my problem has become relative with the setting "ttl" :10. 
Now no more connection problems.

duyuelryel, 2021-01-02:
Thank you for your answer!
But where should i insert in the JSON config? I couldn't find it in the wiki.

DFC1990, 2021-01-02:
Could you Show your Jason ?

duyuelryel, 2021-01-02:
{"c":{"n":"Refi2","l":13,"b":[{"g":0,"t":5}]},"a":[{"t": 22, "g": 14, "n": 3},{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]}]}

DFC1990, 2021-01-02:
Try this:

`{
  "c" : {
    "n" : "Refi2",
    "l" : 13,
    "ttl" : 10,
    "b" : [
      {
        "g" : 0,
        "t" : 5
      }
    ]
  },
  "a" : [
    {
      "t" : 22,
      "g" : 14,
      "n" : 3
    },
    {
      "0" : {
        "r" : [
          {
            "g" : 12
          }
        ]
      },
      "1" : {
        "r" : [
          {
            "g" : 12,
            "v" : 1
          }
        ]
      },
      "b" : [
        {
          "g" : 0
        }
      ]
    }
  ]
}`

duyuelryel, 2021-01-02:
Thank you! I will test it tomorrow at daylight!
I hope, it will help for me too!

Thank you for your help!

ss83, 2021-01-10:
Just a feedback. By now all device start showing the "no respond".
And none is acting within a scene or can be switched using siri. In the app they work (when not showing the no respond). Siri just claims "device does not react".

Any idea what I can check again? Will update all device today to latest firmware (where no ota was available).

closed 2021-02-09 by stale[bot]