RavenSystem/esp-homekit-devices #1278: Gosund SP111

xDuk3x, 2020-12-31:
Hello everybody,

after updating from 3.3.0 to 4.4.3 for the Power measurement update. I Only get the informations for the used Watts, Volts and Amps are showing zero. The Json string is the one from the wiki.

Thanks in Advice
Best Regards
Ronny

WojtekWro, 2021-01-03:
Hi, try update to the last version. 
I have 4.4.6 right now and it works fine. The JSON string from the wiki as well.

closed 2021-02-03 by stale[bot]

lukaszwitkowski, 2021-07-19:
Do you have sp111 with working voltage etc meter ? And in what app do you see that power consupmpcion

lukaszwitkowski, 2021-07-20:
> Hi, try update to the last version. 
> 
> I have 4.4.6 right now and it works fine. The JSON string from the wiki as well. 

Do you have voltage status in home app or only in HAA menager?