RavenSystem/esp-homekit-devices #1280: Magichome one channel dimming issue

gsaplx, 2021-01-02:
Hi,

first of all thanks a lot for the work you’re doing for this. 
I have bought 3 cheap one channel magichome controllers and hook them up to different led strips/christmas lights using this json:
{"c":{"q":850,"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":15,"g":13,"v":12,"w":14,"b":[{"g":0}]}]}

Since I don’t quite understand how the json works and why it works :), I would like to remove the rgb colors from it and leave only the switch and dimming option. 
Another issue is the dimming is flickering for the christmas lights. The dimming of the led strip is okish when you look directly at the lights but when you move your eyes around is a bit disturbing. I hope it makes sense what I’m saying :). 
thanks again,
Gabriel

Hanimaniac, 2021-01-03:
Hey gsaplx,

I use these json for the single LED Controller:

Magic Home WIFI Single LED Strip

{"c":{"q":1000,"b":[{"g":0,"t":5}]},"a": [{"t":30,"w":12,"b":[{"g":0}]},{"1":{"s":[{"a":1}]}},{"t":1,"1":{"s":[{"a":2}]}}]}

With this json, also a Setup-Switch and an Update-Switch are created for better use. (So while Homekit Installation You get One Light and 2 Switches.
Maybe You have to change the gpio's in the json, if your colors of the led stripe don't match with your app colors. 
Greets
Hanimaniac

gsaplx, 2021-01-03:
Thank you Hanimaniac! It’s what I wanted but the christmas lights are still flickering when dimming. Probably the lights don’t support dimming...
Regarding the extra switches, they are really useful. How you organise them in the home app? If you have a lot of lights then the rooms will be full of switches.

Thanks again!
Gabriel

Hanimaniac, 2021-01-04:
I think the flickering while dimming is normal. It's just a cheap led controller. 
I made a 'Setup Room' and a 'Update Room' for all my devices.

gsaplx, 2021-01-04:
Smart and make sense. Thanks a lot for your time! Have a great year :)

serial8killer, 2021-01-14:
> {"c":{"q":1000,"b":[{"g":0,"t":5}]},"a": [{"t":30,"w":12,"b":[{"g":0}]},{"1":{"s":[{"a":1}]}},{"t":1,"1":{"s":[{"a":2}]}}]}

Hey Hanimaniac, what changes do I need to do to the above Json to remove the update and setup switch?

Hanimaniac, 2021-01-14:
Hey Killer 😬,
you can find the normal Json for magichome controler in the HAA Database:
{"c":{"q":850,"b":[{"g":0,"t":5}]},"a":[{"t":30,"w":12,"b":[{"g":0}]}]}
Maybe you habe to change the gpio for your device to work correct.

Hanimaniac, 2024-12-20:
Magic Home WIFI Single LED Strip
 
{"c":{"io":[[[12],7],[[0],6,1]]},"a":[{"t":30,"g":[12]}]}

Magic Home WIFI LED RGBW Strip

{"c":{"io":[[[12,13],7],[[14,15],7,0,0,2]]},"a":[{"t":30,"g":[15,13,12,14]}]}

mafyata, 2024-12-20:
Thanks Hanimaniac. 🙏🏼

mafyata, 2024-12-21:
Next problem. The board is new and there is no text on the pins. Which one is tx,rx,vcc ….?

Hanimaniac, 2024-12-21:
I found this picture on google:

[image]