RavenSystem/esp-homekit-devices #1281: Need Help with Router configuration

sblitzer, 2021-01-02:
Hi All,

Recently I replaced my router with a new one Asus RT-AX88U.
All HAA devices (41) are perfectly working without any issue, but when I want to add a new HAA device (after I already flashed it and config it with JS ) I can't add it to the HK app.
Please see the 
[image]
[image]
[image]

attached configuration of 2.4GHz & 5GHz and pro settings.
Any advice?

Kienz, 2021-01-15:
What error occurs? Which device do you want to pair? JSON of the device? Is it pingable?

sblitzer, 2021-01-15:
I've managed to fix all the issues except one after I perform a firmware update the device never goes online again until I hard reset it (plug off the power and on) or reboot the router.

macjl, 2021-01-15:
Hello, how long did you wait for the device to reconnect after a firmware update?
I’ve noticed same behaviors with some device, but they’re always reconnect after a while, without doing anything. It can take 15 to 20 mn in my case.

sblitzer, 2021-01-15:
They never reconnect again unless I hard reset it or reboot the router.
This issue is definitely router configuration because with my old router this never happened

closed 2021-01-19 by sblitzer