RavenSystem/esp-homekit-devices #1282: Problem with Shelly Dimmer2

skv6, 2021-01-07:
HAA 4.3.4 Shelly Dimmer2 all time in setup mode. Any anothe JSON work, but if I use Dimmer JSON it is in setup mode like wrong JSON. What is my problem?

iotcastro, 2021-01-08:
> HAA 4.3.4 Shelly Dimmer2 all time in setup mode. Any anothe JSON work, but if I use Dimmer JSON it is in setup mode like wrong JSON. What is my problem?

Can you post the JSON to see it?

skv6, 2021-01-08:
JSON from Devices Database
It is OK at json validator

{"c":{"f":20,"r":[{"n":0}],"l":15,"i":0,"b":[{"g":13,"t":5}]},"a":[{"t":30,"b":[{"g":13},{"g":12,"i":1,"p":0,"t":0},{"g":14,"i":1,"p":0},{"g":14,"t":0}],"0":{"u":[{"v":"010202060000f401000000ff04"}]},"y0":[{"v":1,"0":{"u":[{"v":"010202060a00f4010000010904"}]}},{"v":5,"0":{"u":[{"v":"010202063200f4010000013104"}]}},{"v":10,"0":{"u":[{"v":"010202066400f4010000016304"}]}},{"v":15,"0":{"u":[{"v":"010202069600f4010000019504"}]}},{"v":20,"0":{"u":[{"v":"01020206c800f401000001c704"}]}},{"v":25,"0":{"u":[{"v":"01020206fa00f401000001f904"}]}},{"v":30,"0":{"u":[{"v":"010202062c01f4010000012c04"}]}},{"v":35,"0":{"u":[{"v":"010202065e01f4010000015e04"}]}},{"v":40,"0":{"u":[{"v":"010202069001f4010000019004"}]}},{"v":45,"0":{"u":[{"v":"01020206c201f401000001c204"}]}},{"v":50,"0":{"u":[{"v":"01020206f401f401000001f404"}]}},{"v":55,"0":{"u":[{"v":"010202062602f4010000012704"}]}},{"v":60,"0":{"u":[{"v":"010202065802f4010000018b04"}]}},{"v":65,"0":{"u":[{"v":"010202068a02f4010000010904"}]}},{"v":70,"0":{"u":[{"v":"01020206bc02f401000001bd04"}]}},{"v":75,"0":{"u":[{"v":"01020206ee02f401000001ef04"}]}},{"v":80,"0":{"u":[{"v":"010202060032f4010000013104"}]}},{"v":85,"0":{"u":[{"v":"010202065203f4010000015404"}]}},{"v":90,"0":{"u":[{"v":"010202068403f4010000018604"}]}},{"v":95,"0":{"u":[{"v":"01020206b603f401000001b804"}]}},{"v":100,"0":{"u":[{"v":"01020206e803f401000001ea04"}]}}]}]}

iotcastro, 2021-01-08:
I have been try this JSON with version 4.5.1. and runnig correctly with a ESP8266, but you can try add "w":5.

{"c":{"f":20,"r":[{"n":0}],"l":15,"i":0,"b":[{"g":0,"t":5}]},"a":[{"t":30,"w":5,"b":[{"g":13},{"g":12,"i":1,"p":0,"t":0},{"g":14,"i":1,"p":0},{"g":14,"t":0}],"0":{"u":[{"v":"010202060000f401000000ff04"}]},"y0":[{"v":1,"0":{"u":[{"v":"010202060a00f4010000010904"}]}},{"v":5,"0":{"u":[{"v":"010202063200f4010000013104"}]}},{"v":10,"0":{"u":[{"v":"010202066400f4010000016304"}]}},{"v":15,"0":{"u":[{"v":"010202069600f4010000019504"}]}},{"v":20,"0":{"u":[{"v":"01020206c800f401000001c704"}]}},{"v":25,"0":{"u":[{"v":"01020206fa00f401000001f904"}]}},{"v":30,"0":{"u":[{"v":"010202062c01f4010000012c04"}]}},{"v":35,"0":{"u":[{"v":"010202065e01f4010000015e04"}]}},{"v":40,"0":{"u":[{"v":"010202069001f4010000019004"}]}},{"v":45,"0":{"u":[{"v":"01020206c201f401000001c204"}]}},{"v":50,"0":{"u":[{"v":"01020206f401f401000001f404"}]}},{"v":55,"0":{"u":[{"v":"010202062602f4010000012704"}]}},{"v":60,"0":{"u":[{"v":"010202065802f4010000018b04"}]}},{"v":65,"0":{"u":[{"v":"010202068a02f4010000010904"}]}},{"v":70,"0":{"u":[{"v":"01020206bc02f401000001bd04"}]}},{"v":75,"0":{"u":[{"v":"01020206ee02f401000001ef04"}]}},{"v":80,"0":{"u":[{"v":"010202060032f4010000013104"}]}},{"v":85,"0":{"u":[{"v":"010202065203f4010000015404"}]}},{"v":90,"0":{"u":[{"v":"010202068403f4010000018604"}]}},{"v":95,"0":{"u":[{"v":"01020206b603f401000001b804"}]}},{"v":100,"0":{"u":[{"v":"01020206e803f401000001ea04"}]}}]}]}

skv6, 2021-01-09:
Thank You. 
HAA 4.3.4 and some other work only add "w":5
HAA 4.5.1 work without it.

closed 2021-02-09 by stale[bot]