RavenSystem/esp-homekit-devices #1286: Massive HAA update.

vsevolod-volkov, 2021-01-09:
Hi, Jose!

I have got semi-deployed HimeKit-based system currently using above 40 HAA components. The expected HAA component amount is about hundred after deployment completion.

Every time I decide to upgrade HAA firmware I walk through my home and switching every of component button 8 times to enter setup mode result initiation of automatic OTA update after 90 seconds.

When all the components are updated, I inspect all of them with My Home application to prove if version is successfully updated.

That takes a lot of time for every update.
**So, my question: what massive update strategy do you recommend for relatively big HAA installations.**

Thanks.

pilot1981, 2021-01-09:
You can do it in this way:
1) only once enable OTA update on setup
Mode of every device
2) add a dummy switch in every HAAD device to enter in setup mode
3) group all dummy switch all together

To update all devices you need only to activate this group of dummy switches

vsevolod-volkov, 2021-01-09:
@pilot1981 thank you for your reply. Will try that.

vsevolod-volkov, 2021-01-11:
It works.

closed 2021-01-11 by vsevolod-volkov