RavenSystem/esp-homekit-devices #1287: Flashing MagicHome RGBW without GPIO?

brymonster, 2021-01-11:
Hello HAA users.
recently received another MagicHome RGBW led controller. unfortunately, on cracking the case to flash, the usual GPIO is missing. instead, there is a ["boot"](https://imgur.com/a/lTCSuc7) solder pad. I tried using this as I would with GPIO (connect  to GND as booting with FTDI) but it has not worked yet. I also tried using both V33 and GND. no joy. 

anyone else come across this yet?

duyuelryel, 2021-01-11:
There is an RXD and TXD pin, but you must to check if the ic is annesp, because there is an other version of controller with some strange ic what is not compatible with tasmota!

brymonster, 2021-01-11:
You are absolutely correct. [It does NOT look like an esp chip.](https://imgur.com/a/lTCSuc7) 
Wow this is going to make it hard to buy MagicHome going forward!

duyuelryel, 2021-01-11:
Yes. I got the same from ali. But it's still working nicely and i can make it work with this node. I'm not sure if HA or anything else can discover it.

closed 2021-01-12 by brymonster

smibrandon, 2022-11-12:
I have realized I have the EXACT SAME controller. Have you found a way to control it directly from HA in the past year-and-a-half, by any chance?