RavenSystem/esp-homekit-devices #1288: Bug supposed: fan accessory (t: 65) ignores notifications sent by other accessory

vsevolod-volkov, 2021-01-11:
HAA v4.5.1, v4.6.1, other version not tested.

v:1 notifications are ignored although v:0 notification produces corresponding 0-action well (fan stop).

```
"a": [
    {
...
    },
    {
      "t": 65,
      "sn": "FAN",
      "0": { "r": [ { "g": 15 } ] },
      "1": { "r": [ { "g": 15, "v": 1 } ], "m": [ { "g": 3, "v": -1 } ] }
    },
    {
      "d": 1200,
      "sn": "FAN-TIMER",
      "0": { "m": [ { "g": 2, "v": 0 } ] },
      "1": { "m": [ { "g": 2, "v": 1 } ] }
    },
...
  ]
```

RavenSystem, 2021-01-11:
It is not a bug:
`0` is off.
`1` to `100` set fan target speed.
`101` is on.

vsevolod-volkov, 2021-01-11:
Thank you for fast reply. Have already tested as you suggested - it really works.
May be better to reflect that in accessory 65 notification doc?
[image]

RavenSystem, 2021-01-11:
Yes, I see it. Thanks!

RavenSystem, 2021-01-11:
Done!

closed 2021-01-11 by RavenSystem