RavenSystem/esp-homekit-devices #1292: Unresponsive after a few hours

serial8killer, 2021-01-13:
Hey Guys.

First of all I would like to thanks all the hard work put in this project by everyone, this is a great way for thinkers to convert cheap chinese devices to homekit!

So, I have several LED strip controllers and I've flashed 2 controllers yesterday, the controller in question is the one in the image bellow, after flashing and configuring they work great, with the JSON config  bellow, but for some reason a few hours after this they get unresponsive on Homekit... I have to unplug them from the wall and plug them on and then they work for a few hours again then they get unresponsive again... As you can tell like this is not usable for automations or even day to day life. Please help me out, ask whatever you need.

[image]

JSON config:
{"c":{"l":2,"b":[{"g":13,"t":5}]},"a":[{"t":30,"r":15,"g":12,"v":14,"w":4,"b":[{"g":5,"p":0},{"g":5,"t":0},{"g":13}]}]}

Kienz, 2021-01-15:
Which HAA version do you use? Update to the newest 4.7.2.
Which WiFi Setup do you have? How many WiFi devices? etc.
Any problems with other WiFi HK devices?

serial8killer, 2021-01-18:
Sorry for late reply.

I'm not sure of the version it's on but since I've flashed the controller 6 or 7 days ago, I was on the most recent build, although I see now that there was an update yesterday with several network improvements, so I will try to update.

I have a good main Netgear access point and another wifi powerline repeater.

The controller was set up in the main network and then brought to the bedroom with the repeater is.

The network mode is set to normal, in normal mode the controller is going to roaming to the new closer wireless point, or is going to forcefully keep trying to connect to the main network?

In either way, 2.4ghz has a very good signal even in the bedroom, so the controller was working for a full day or 2 before starting to give problems.

Several wifi devices, although none give problems, including another same model controller also with the same version (flashed at the same time) of the HAA that works perfectly

closed 2021-02-17 by stale[bot]