RavenSystem/esp-homekit-devices #1296: Garage force opens after power outage

Saurman, 2021-01-14:
Hello, I have Sonoff v1 configured to act as garage door opener with 2 end switches connected to GPIO 4 and 5. 

The issue is that when power (along with wifi) goes down, on power-on gate force opens and relay remains ON until I change garage state via HomeKit app (clicking garage actually does nothing but it releases relay and I can operate door using my wall switch) 

My config:
`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2},"t":40,"s":5,"d":20,"f2":[{"g":4,"t":1,"i":1,"p":1}],"f3":[{"g":5,"t":1,"p":1,"i":1}],"b":[],"f4":[{"g":5,"t":0,"i":1,"p":1}],"f5":[{"g":4,"t":0,"p":1,"i":1}]}]}`

Saurman, 2021-01-14:
tried with S:1 on v4.7 - same behaviour. When power is removed garage door opens which is obviously not good for security

rjblake, 2021-01-14:
Did you try add `"0":{"r":[{"g":12}]}` in the config options? Also, perhaps check this [thread](https://github.com/RavenSystem/esp-homekit-devices/issues/1226#issuecomment-733624169).

Saurman, 2021-01-14:
@rjblake thanks a lot, missed that one, worked like charm!

closed 2021-01-14 by Saurman

djsomi, 2021-02-09:
Guys, can someone explain what changed in the FW? I also had to add the part to pull gpio12 to low, but I want to understand why? 
What is changed?

Thanks!