RavenSystem/esp-homekit-devices #1297: Garage Door stoped working properly in last update to 4.7.2 - it's stuck on "Opening..."

tlagarto, 2021-01-15:
I'm using a Sonoff Mini modified as a dry contact with a magnetic sensor to activate the "Closed" state.
It has been in use as a Garage Door for one year working flawlessly.

Every now and then I update the firmware, but this one has been on an earlier version for a while version (maybe 4.4.x or even earlier).
Since the last update to 4.7, then to 4.7.1, and yesterday to 4.7.2, it stoped working properly and gets stuck on "Opening..." state.
[image]

If I reboot it, it will assume the "Closed" state (if closed) or "Open" (if in any other position).
As soon as I operate the garage (either via Home app or manually or legacy remote control) it gets stuck on the "Opening..." state forever.

This is the Json code I'm using:
`{"c":{"0":{"r":[{"g":12}]},"l":13,"b":[{"g":0,"t":5}]},
"a":[{
  "0":{"r":[{"g":12,"v":1,"i":0.5}]},
  "1":{"a":0},
  "2":{"a":0},
  "3":{"a":0},
  "t":40,
  "d":15,
  "xa":0,
  "f3":[{"g":4,"t":0}],
  "f4":[{"g":4,"t":1}],
  "s":0,
  "b":[{"g":0,"t":1}]},
  {"1":{"s":[{"a":1}]},"i":1,"b":[],"s":0},
  {"1":{"s":[{"a":2}]},"t":2,"i":0.5,"b":[],"s":0}]}`

The last 2 buttons are virtual ones to get the device on setup mode and fw update mode.
They also become non-operational when the device is stuck on "Opening..." state.

Flashed the device 3 times already and it always reverts to the same behaviour...
I do not keep previous versions FW, so I can't test it with something prior to 4.7.

Is anyone experiencing the same? Is there a fix or am I doing something wrong?

shanghaix, 2021-01-15:
Did you have a look at #1296

tlagarto, 2021-01-15:
I did. But that is not the same issue since the door does not open on power on.
I’ve tried a combination of strings including the one mentioned.
Thank you, but the issue persists...

rjblake, 2021-01-17:
> I'm using a Sonoff Mini modified as a dry contact with a magnetic sensor to activate the "Closed" state.
> ...
> I **do not keep previous versions FW**, so I can't test it with something prior to 4.7.
> 

All previous releases can be found on the Release page - [here](https://github.com/RavenSystem/haa/releases)

closed 2021-02-17 by stale[bot]