RavenSystem/esp-homekit-devices #1298: Wi-Fi Reconect failure

Drumerty, 2021-01-16:
Hello, I am having a problem with two of my three Sonoff devices.
I live in the private sector where power outages are quite common. 
After blackout, when the electricity turns on again and by the time the Wi-Fi router is loaded(Mi Router 3), only one device works in the Home App (**Bassic R3 with open sensor**), the rest (**Sonoff Mini (Door Lock type) and Sonoff TH-10(Thermostat**) do not response  even after an hour), furthermore i am unable to ping them. 
They start working only after i unplug and plug them back.

Please help with this problem.

Kienz, 2021-01-17:
Sometimes I have the same problem with my Shelly 1/2.5 devices. I reconnect WiFi (over UniFi Controller) - they are displayed but aren't pingable. Only after unplug and plug back they work.

Kienz, 2021-01-18:
Also happens with 4.7.3 and 4.7.4

Drumerty, 2021-01-18:
> Also happens with 4.7.3 and 4.7.4

For me too.

My problem is that the Sonoff device starts working before my router starts bssid streaming.
And this factor is causing this problem.
Is there a way to delay Sonoff from searching for the network on startup?

RavenSystem, 2021-01-25:
Wi-Fi reconnection is fixed with v4.8.0 and above.

closed 2021-01-25 by RavenSystem