RavenSystem/esp-homekit-devices #1299: Inverted PWM Logic - possible?

Fafex, 2021-01-17:
Hello,

I've got a problem with my self made PWM dimmer. I was my own "Arduino made" MQTT software but I would love to switch to HAA. 

The problem is that mosfet driving the LED strip is driven by another mosfet. Schematic attached
[image]

Here's my config:

`{"c":{"o":1,"l":13,"i":0,"q":850,"b":[{"g":0,"t":5}]},"a":[{"w":4,"t":30,"s":0}]}`

But now the off state gives me 12V on the output, and 100% on gives me 0V.

Is there a way to invert the PWM logic or will I have to do some soldering?

Thanks for any reply.

RavenSystem, 2021-01-25:
Currently, no. But v5.0 will have that option.

RavenSystem, 2021-02-18:
Now you can declare PWM GPIOs with inverted logic.
https://github.com/RavenSystem/esp-homekit-devices/wiki/Lightbulb#RGBW-GPIO-Channels

closed 2021-02-18 by RavenSystem