RavenSystem/esp-homekit-devices #1302: Wifi mode explanations

serial8killer, 2021-01-19:
I'm sorry if this is already posted here somewhere, but I cannot find it.

Could someone please explain what are the differences between all the wifi modes?

Some documentations would be great so people can choose accordingly to their use case and network configuration.

pawelosin, 2021-01-25:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode#change-wifi-network

Passive roaming: device will scan for Wifi networks at boot and when it disconnects, then it will connect to BSSID with best RSSI.
Active roaming: same as Passive roaming; and every 35 minutes, device will scan for Wifi networks and it will reconnect to BSSID with best RSSI when it is connected to other. HAA will be unresponsive over Wifi about 2 seconds when it is performing a Wifi scanner.

macjl, 2021-01-25:
But what is the difference between "passive roaming" and "normal"?
It’s not clear for me how the "normal" mode choose it’s Wifi AP.

wrx5, 2021-01-27:
and also what the WiFi Captive Portal is?

sorriso93, 2021-02-04:
> and also what the WiFi Captive Portal is?

it is the setup page you load when the [device is in setup mode](https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode)

sorriso93, 2021-02-04:
> But what is the difference between "passive roaming" and "normal"?
> It’s not clear for me how the "normal" mode choose it’s Wifi AP.

from the wiki 

> The WiFi connection mode button can be toggled between Normal and Force BSSID. Choosing Normal will cause the device to connect to any network with the selected SSID. Choosing Force BSSID will force the device to connect to the WiFi network with the specified BSSID only i.e. if you have range extenders in your house all with the same SSID you can use the Force BSSID option to ensure the device only connects to a specific range extender.

macjl, 2021-02-04:
So in normal mode, the BSSID is choosen totally randomly?

peros550, 2021-02-04:
I think in normal mode the device looks for the BSSID irrespective of which channel that BSSID operates. In Forced mode, it looks for BSSID+Channel that you have selected.

RavenSystem, 2021-02-26:
Wiki is updated.

closed 2021-02-26 by RavenSystem