RavenSystem/esp-homekit-devices #1304: shelly1 bootloop

borissov, 2021-01-22:
Hello, 
I install HAA on Shelly1 but get bootloop :( 

My flash settings:

[image]

This is serial log:

[image]

Where am I wrong? I think the bootloader is not original from shelly? Thanks in advance!

Kristian8606, 2021-01-22:
Подозирам, че пишете в паметта, където не би трябвало, това може да бъде причинено от невалидни указатели.

borissov, 2021-01-22:
> Подозирам, че пишете в паметта, където не би трябвало, това може да бъде причинено от невалидни указатели.

//In Bulgarian: След загубени няколко часа, вероятно става въпрос за проблем със захранването на чипа, но все още не съм го оправил проблема и вече малко се изнервих. Оказва се че или не записва като хората в чипа или не може да буутне. Тъй като нямам друг uart адаптер няма как да тествам, сега съм го закачил на лабораторно захранване, което по принцип е доста точно и тока няма много пулсации и си буутна.

Kristian8606, 2021-01-22:
Захранването беше следващото ми предложение :)

closed 2021-02-22 by stale[bot]