RavenSystem/esp-homekit-devices #1306: Flashed the sonoff mini with latest fullhaaboot.bin, when joining the WiFi doesn't show captive portal

aleroot, 2021-01-23:
I have flashed a sonoff mini with the latest fullhaaboot.bin found the in installation guide of of this repository, when accessing the WiFi the iPhone doesn't show the captive portal anymore... 

What is the port the setup is listening to ? 
I can try to access the web portal manually ..

iPhone version 14.3

mariusvasss, 2021-01-23:
http://192.168.4.1:4567/

mariusvasss, 2021-01-23:
Good Luck!

aleroot, 2021-01-23:
Just noticed in the release notes of the latest version, that the captive portal has been disabled ... with no apparent reason ...

This project is now releasing updated very often and very often breaking features or device previously working correctly ... 

I don't understand the way this project is led, unbelievable. 

Would be interesting to know why the captive portal has been removed. I'm gonna probably create my own fork from now own, the project is highly unstable, so it is not trustable...

closed 2021-01-23 by aleroot