RavenSystem/esp-homekit-devices #1308: WiFi disconnecting in version 4.9.0 after a while

batjoc, 2021-01-24:
Currently I'm experiencing some problems with version 4.9.0 on a son off mini and a shelly 1. I've attached the logs of the Shelly 1. Both disconnect after a while and are not reachable anymore. I had some connection problems with version 4.6.x as well, but version 4.8.x was running really stable. However with version 4.9.0 there are again some wifi problems.

[session.log](https://github.com/RavenSystem/esp-homekit-devices/files/5861975/session.log)

aleroot, 2021-01-25:
Probably you better downgrade to 3.x version, because latest versions are very unstable unfortunately. There are weekly updates reporting improved Wi-Fi, but rather than improvements are just twiddling around changes, without any clear view of what to do to really fix the issue or improve... The typical: EDIT AND PRAY kind of changes.

I have still some devices on version 1.x and they are the most reliable. I suggest you to not go above v3.5.17 if you don't really need any of the new features, from my point of view is one of the most stable recent versions.

NewTris, 2021-01-25:
We checked some SonOff Basic / 4CH or Wemos Mini with the new Version 4.9.0 and there is no way for getting the reconfiguration again (to check the JSON Config) about puhing 8 / 10 time the bulb or switch. It doesn't react .. is there a new way to push the config back ? Or is this a Bug ?

serial8killer, 2021-01-25:
NewTris, the captive portal is gone, now you have to go to the ip of the device together with the port 4567, for example http://192.168.4.1:4567/

batjoc, 2021-01-25:
Seems it's running a lot more stable lately. Not sure what was causing it, but closing this issue for now

closed 2021-01-25 by batjoc