RavenSystem/esp-homekit-devices #1310: HAA GarageDoor without any sensor (just time), issue

HerwarthHK, 2021-01-24:
Hello guys, totally newbie here :).
Please, I have continuous issues with JSON code for our shared garage door - its “shared” so I cant place magnet sensors or optical obstruction sensor there. So I installed Sonoff SV with HAA to just trigger dry relay connected to my RF remote. So in this situation I just need to JSON to control the garage door with spcified total time of opening/closing duration (to see animation of position). I have tried almost everything but with every try the garage stuck in “opening” until I press it again. I have found some others guy solution to make “dummy switch” and automation (see screen). I have read that RavenCore supported this but new one HAA cant automatically close the door :-(. Is that true or is there some other easier solution? Please again - no sensors, just time! Thanks in advance. David
[image]
[image]

mariusvasss, 2021-01-24:
The garage door needs magnet to know the status of the door. You can leave it as a switch only with itching. Apple configuration for garage doors needs at least one sensor (magnet) for working.

samialdury, 2021-01-29:
@HerwarthHK 
Hi, I had a similar problem. See my solution [here.](https://github.com/RavenSystem/esp-homekit-devices/issues/1266#issuecomment-749836860)

anto-toto, 2021-02-21:
hi, i want to use a low voltage sonoff to control my garage.  I use a low voltage sonoff to connect it directly to the garage control unit.  I need help with the Json string.  It is possible to set the opening of the grage so that at the opening command the sonoff sends an impulse to the control unit button and after twenty seconds it sends another one to block the opening.  and then at the closing command you send a single impulse to close the garage.

mariusvasss, 2021-02-21:
Garage Door no sensors

{
  "c":{"l":13,"b":[{"g":0,"t":5}],"0":{"r":[{"g":12}]}},
  "a":[{
    "t":40,
    "d":18,
    "0":{"r":[{"g":12,"v":1,"i":0.5}]},
    "1":{"a":0},
    "2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},
    "3":{"a":2}
  }]
}

anto-toto, 2021-02-21:
ciao, se ho capito bene questa stringa è impostata con un'apertura di 18 secondi e chiusura anche di 18 secondi. io vorrei che al completamento dell'apertura venga dato un altro impulso per arrestare l'apertura del garage. in quanto utilizzo il sonoff per aprire una serranda molto alta e vorrei evitare di aprirla tutta.

Saluti
Antonio Vitrò

Da: mariusvasss
Inviato: domenica 21 febbraio 2021 11:24
A: RavenSystem/esp-homekit-devices
Cc: anto-toto; Comment
Oggetto: Re: [RavenSystem/esp-homekit-devices] HAA GarageDoor without anysensor (just time), issue (#1310)

Garage Door no sensors
{
"c":{"l":13,"b":[{"g":0,"t":5}],"0":{"r":[{"g":12}]}},
"a":[{
"t":40,
"d":18,
"0":{"r":[{"g":12,"v":1,"i":0.5}]},
"1":{"a":0},
"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},
"3":{"a":2}
}]
}

anto-toto, 2021-02-27:
hi, if I understand correctly this string is set with an opening of 18 seconds and also closing of 18 seconds.  I would like another impulse to be given upon completion of the opening to stop the garage opening.  as I use the sonoff to open a very high shutter and I would like to avoid opening it all.

closed 2021-03-31 by stale[bot]