RavenSystem/esp-homekit-devices #1313: Shelly uni flash

pasparakis, 2021-01-25:
Hi everyone!!! has anyone flashed this device?  any available Json string?[image]
[image]

https://shelly.cloud/products/shelly-uni-smart-home-automation-device/

macjl, 2021-01-25:
Hello. I did. Here is my JSON for a window opener with setup mode button. 

`{"c":{"z":0,"o":3,"l":0,"i":0,"0":{"r":[{"g":4},{"g":15}]}},"a":[{"t":45,"w":1,"o":18,"c":14,"0":{"r":[{"g":15,"v":1},{"g":4}]},"1":{"r":[{"g":15},{"g":4,"v":1}]},"2":{"r":[{"g":15},{"g":4}]},"3":{"r":[{"g":15,"i":0.2},{"g":4}]},"4":{"r":[{"g":15},{"g":4,"i":0.2}]}},{"t":1,"1":{"s":[{"a":1}]}}]} `

pasparakis, 2021-01-25:
Thank you friend!!!

pasparakis, 2021-01-25:
any json string for doorbell?  has anyone tried it?

closed 2021-02-25 by stale[bot]

ra1nb0w, 2021-04-11:
Probably the json example of @macjl can be posted to the database's wiki page.