RavenSystem/esp-homekit-devices #1314: Tuya Smart Motorized Roller Blinds with HAA?

pawelosin, 2021-01-25:
Is there a way to use this Tuya WiFi M515EGWT Smart Motorized Roller Blinds with HAA?
https://www.aliexpress.com/item/1005001890929280.html?spm=a2g0s.9042311.0.0.69d74c4dyM8Hy3
Inside it looks like this 
[image]
[image]
[image]
[image]

closed 2021-02-25 by stale[bot]

escobarin3, 2022-01-17:
Works with the Tuya Integration or Local Tuya, withouth issue. At least it works for me. I have 4 of them

cybercorey, 2022-05-22:
I know this is stale, but I have these now, I've managed to flash tasmota on it no worries and can tell it to go up and down and stop. However I cant seem to set the limits.

thomascandrian, 2022-07-24:
Have the same, did work with the app for ~2 days. Now it's kind of dead. Wnat to see if I can flash it with tasmota. @cybercorey can you tell which settings you did use on tasmota, i.e. the template?

cybercorey, 2022-07-24:
Hi @thomascandrian Ended up following this guide: https://blakadder.com/tuya-covers/ which covers getting tasmota on there and home assistant setup.

I had to work out a couple of the MCU mappings myself however:
```
0x55AA 00 06 0005 10 04 00 01 04 23 Remove Limits SerialSend5 55AA00060005100400010423
0x55AA 00 06 0005 10 04 00 01 02 21 Remove Upper Limit SerialSend5 55AA00060005100400010221
0x55AA 00 06 0005 10 04 00 01 03 22 Remove Lower Limit SerialSend5 55AA00060005100400010322

0x55AA 00 06 0005 10 04 00 01 01 20 Set Lower Limit SerialSend5 55AA00060005100400010120
0x55AA 00 06 0005 10 04 00 01 00 1F Set Upper Limit SerialSend5 55AA0006000510040001001F
```

thomascandrian, 2022-08-18:
Finally had time to test it, and it works flawlessly. Important: As long Tasmota is not configured as Tuya MCU, nothing works (seems that the MCS waits for a heartbeat form Tasmota).
My Template is
`{"NAME":"TYWE3S in ZS Cover","GPIO":[1,2272,1,2304,1,1,0,0,1,1,1,1,1,0],"FLAG":0,"BASE":54}`

The rest is described in the link from cybercorey - awesome!

escobarin3, 2022-09-11:
Anyone know where can I buy only the motor? Mine stopped working... 

[image]

thomascandrian, 2022-09-13:
@escobarin3 I would bet that it is not the engine that does not work but the stupid TuyaMCU that stops working. 

After happen 2 times, I decided to flash tasmota on the ESP and reconnect the important functions directly to the ESP instead of the stupid tasmota sh**.

* Connect the control pins of the motor driver IC (below the daughterboard on picture 1 above, just to the right of the motor connector) to the GPIOs of the ESP. Disconnect these connections to the TuyaMCU on the board, or you may run into problems later.
* I also connected the two LEDs (blue and red, again disconnect them from the TuyaMCU) and the up and down buttons to the ESP.
* Be careful not to use GPIO0 and GPIO2, as they must remain logically connected or the ESP will not boot.
* I used the following commands on the Tasmota to use it as a curtain, which is automatically detected in Home Assistant. It is based on timing, so it is not very precise, but for me the precision is sufficient.

You can see in the template editor which GPIO I used for which function. 

```
Template {"NAME":"TYWE3S ZS Cover   Tasmota","GPIO":[1,1,32,1,256,257,1,1,576,288,33,1,1,1],"FLAG":0,"BASE":18}}     Module 0
Interlock 1,2
Interlock On
SetOption80 1
ShutterMode 1
ShutterOpenDuration 76
ShutterCloseDuration 72
```

Last change was that I changed LED_LINK to LEG_LINK_i so that the LED doesn't interfere with my sleep :-)

In my opinion a good device with terrible electronics.

escobarin3, 2022-09-13:
@thomascandrian thanks for your reply, but in my case the motor is the problem. I have 4 more Tuya Smart Motorized Roller Blinds so I swapped the motors to see if the issue was the motor, and it was. :/

pawelosin, 2022-11-19:
> Anyone knows where can I buy only the motor? Mine stopped working...
> 
> [image]

Did you find the motor? I have the same issue.

Wotty, 2023-04-10:
Also looking for motor, closed I found was this: https://m.alibaba.com/product/1600254336945/24mm-mabuchi-3V-3-7V-4.html

But has a min order qty of 5. Meaning even getting 4 of the cheapest £.3 and one 29Z370 it would cost £20 eith shipping. Hence I've ordered a new one for £40.

phillipse01, 2023-10-29:
I got a set of these and they are now rocking tuya WBR3's which I understand is not flashable - looking to brain swap replace with ESP32-C3F

frostworx, 2023-12-09:
I can confirm what @phillipse01 mentioned.

Just received 2 devices and they have a WBR3 chip now.
As `libretiny` support was recently added to esphome ([esphome wiki](https://esphome.io/components/libretiny.html)), there's hope that we can flash esphome on them ${one_day} anyway. Found this recent discussion snippet which sounds promising:

https://pay.reddit.com/r/Esphome/comments/17fpzm0/tuya_mcu/k6brq6u/

```
You probably could have flashed esphome onto the wbr3 module with libretiny.
But anyway, make sure you have the right gpio pins...

Not quite yet, currently only 87x0B is fully supported by LibreTiny. WBR3 is 87x0C, which isn't fully ready yet (no wifi or OTA), still work in progress.
```

frostworx, 2024-11-28:
I haven't tested this yet, but this [recent comment](https://github.com/libretiny-eu/libretiny/issues/44#issuecomment-2506125757) in the [libretiny upstream issue](https://github.com/libretiny-eu/libretiny/issues/44) looks very promising.

phillipse01, 2024-11-29:
I was able to solve this issue by brain swapping the wbr3 with an esp32cf and flashing esphome and use the "Tuya cover" option