RavenSystem/esp-homekit-devices #1316: how to make the switch auto turn off after 1 second, using for a door lock please help

mahdy89, 2021-01-26:
how to make the switch auto turn off after 1 second, using for a door lock please help

mariusvasss, 2021-01-26:
here is the JSON

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":1,"i":1}]},"1":{"r":[{"g":12,"v":0}]},"t":4,"b":[],"s":1,"i":1}]}

i use it on a sonoff mini, you have to change your "g" number if its another device

mahdy89, 2021-01-26:
thank youuuu, can we make it unlock twice? my door dont open from first try

mariusvasss, 2021-01-26:
Extend your time, you want 2 pulses?

mahdy89, 2021-01-26:
yes i want 2 pulse

mahdy89, 2021-01-27:
are you able to help me with that please? i will be verry thankfull

mariusvasss, 2021-01-27:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":1,"i":1},{"g":12,"v":1,"i":1}]},"1":{"r":[{"g":12,"v":0}]},"t":4,"b":[],"s":1,"i":3}]}

i think this is it. try an confirm!

mahdy89, 2021-01-27:
tried it.. didnt work :(

mariusvasss, 2021-01-27:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12,"i":1},{"g":12,"v":1,"i":1},{"g":12,"v":1,"i":1}]},"1":{"r":[{"g":12,"v":0}]},"t":4,"b":[],"s":1,"i":1}]}

if this doesn't work, i don't know.....

mariusvasss, 2021-01-28:
It worked?

closed 2021-02-28 by stale[bot]