RavenSystem/esp-homekit-devices #1319: Tuya Devices

MarkM500, 2021-01-27:
Hi,

is it possible to flash Tuya Devices, like Door sensors, leak sensors or fire/smoke sensors?
like this one fror example: https://de.aliexpress.com/item/1005001622897859.html?spm=a2g0s.9042311.0.0.27424c4dWHObjA or this one:
https://de.aliexpress.com/item/4001077711202.html?spm=a2g0s.9042311.0.0.27424c4d2OVSqD

Inside view:
[image]

Often the MCU is also working, not only the ESP8266?! How to find out, which GPIO is taken?

Thanks for your ideas.
Would be nice, if something like this is possible.
Mark

morphix9, 2021-02-08:
What about https://www.npmjs.com/package/homebridge-tuya ?

MarkM500, 2021-02-09:
Hi thanks for your answer, but it is not possible to integrate "Sensors" in home bridge-tuya, because the problem, that these sensors give only signal with the event and after that they going to sleep.

Mark

tonton-RISK, 2021-02-13:
> Hi thanks for your answer, but it is not possible to integrate "Sensors" in home bridge-tuya, because the problem, that these sensors give only signal with the event and after that they going to sleep.
> 
> Mark

Hi Mark,

Look this
[[image]](url)

[[image]](url)

json : {"NAME":"TUYA SENSOR","GPIO":[0,107,0,108,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":54}

branigan, 2021-03-12:
I've flashed multiple tuya devices using tuya convert https://github.com/ct-Open-Source/tuya-convert

pilot1981, 2021-03-12:
@branigan also tuya devices with lastest firmware? which workaround do you used for it?

branigan, 2021-03-12:
I flash the firmware about 3 months ago. I don't know version firmware. 
I'use a laptop with ubuntu with wifi as access point. 
I've flashed 2 outlets , 2 switches and 2 outlets with leds (maxcio)

PS: If you want to use door sensor with HomeKit devices you can use Xiaomi sensor with v3 gateway (wifi, zigbee, and bt) that are native compatible with HomeKit from Xiaomi app. (Recently I've buy two and works flaw sly)
door sensor:
https://de.aliexpress.com/item/32749227652.html
gateway:
https://de.aliexpress.com/item/4000493652299.html

closed 2021-04-16 by stale[bot]