RavenSystem/esp-homekit-devices #1320: Revert to stock or OTA flash alternative F/W ??

healeydave, 2021-01-27:
Hi,
Apologies if this is not the right place but someone on discord advised me to post on the GitHub and I'm not sure if this qualifies as an issue or a feature request. I couldn't find a feature request section, hence posting here.

I have a Shelly 2.5 that I (foolishly) accidentally installed HAA. I know, how is that possible you may ask but I was scrolling through https://github.com/yaourdt/mgos-to-tasmota/releases/tag/release-v0.4.6 and somehow I chose the wrong firmware, I'm still baffled myself.

Anyway to cut a long story short, I would like to OTA back from HAA to Shelly stock firmware and so far I haven't been able to find a way to do it. I've never come across a system before that allows you to only go one way with OTA and no way back but the more I look the more this looks to be the case for HAA. 

If this is indeed true, I think the developer really should reconsider this. 

There are some python scripts with multi-stage processes (where necessary) to apply OTA firmwares so I'm hoping there should be something possible to revert the process:
e.g:
https://github.com/yaourdt/mgos-to-tasmota/blob/main/tools/flash-shelly.py

TIA

RavenSystem, 2021-01-28:
Sorry, for security reasons, it is not possible changing firmware using OTA. HAA OTA Installer can only install oficial signed firmwares to prevent unwanted installations of malicious or hacked files.

closed 2021-01-28 by RavenSystem

healeydave, 2021-01-28:
What if the unwanted installation is HAA. This is totally unacceptable, you are effectively bricking peoples devices into one specific firmware if they don't have the tools / ability to get access to the physical programming pins.

ssaavedra, 2021-02-12:
> Sorry, for security reasons, it is not possible changing firmware using OTA.

I believe this stance should be reconsidered.

Why is it a security concern if it requires physical access to the device in order to access Setup Mode?

What's your threat model? If an attacker has already got physical access to your device, they can just swap it for a malicious one.

How does preventing flashing improve security in any aspect?

healeydave, 2021-02-13:
Agreed, I could just ignore this thread now since I have re-flashed (wire-flashed) the device in question and will never be going anywhere near this firmware again for the above reason but I feel strongly enough about the position I was put in, to still advocate support for OTA flashing since some other people in the future may not be so fortunate to remove this firmware as I was able to.

csc-solution, 2022-09-26:
H,

i think so too, it's abs olutly unacceptable. We have more than 50 LED devices and there is no possibility to open it without destroying. It cannot be the point of open source software that after flashing there is a firmware lock and a rollback is no longer possible. Everybody withe in the same situation have to throw all devices in the trash and lose a lot of money.

I hope for an early solution to finally free all devices from the HAA firmware. The easiest is a HAA2Tasmota update, just as easy as Tasmota2HAA.

Best regards,
Thomas

daryel, 2023-12-22:
Would be nice if there was a "signed" version of tasmota-lite.bin which we could flash over HAA.  Seems like everyone is going to Tasmota first to install HAA, would be nice to have a step to back things out.  I don't see it as a security issue, more like a 'locked in' issue.  

This is like telling people you HAVE to install Windows on your laptop and by installing Linux you are now locked into running Linux forever unless you force a user to open their laptop and flash the BIOS with an EEPROM programmer.

RavenSystem, 2023-12-23:
To clarify some details:

> Why is it a security concern if it requires physical access to the device in order to access Setup Mode?

It is false. Setup mode can be accessed without physical access to the device.

> What's your threat model? If an attacker has already got physical access to your device, they can just swap it for a malicious one.

Again, setup mode can be accessed without physical access to the device. Please, read the Wiki to know how it works.

> How does preventing flashing improve security in any aspect?

Preventing OTA flash from malicious firmwares, because setup mode can be accessed without physical access.

> Would be nice if there was a "signed" version of tasmota-lite.bin which we could flash over HAA. Seems like everyone is going to Tasmota first to install HAA, would be nice to have a step to back things out. I don't see it as a security issue, more like a 'locked in' issue.

Installing other firmware from Tasmota is a Tasmota feature. HAA has not this feature. I could sign a tasmota-lite.bin file, however, this is useless because HAA has not any needed mechanism to flash the bootloader part, and there is not enough free flash storage to code all needed instructions (many kilobytes of space are needed). There is not any 'locked in', you can flash your device with a wired connection.

> This is like telling people you HAVE to install Windows on your laptop and by installing Linux you are now locked into running Linux forever unless you force a user to open their laptop and flash the BIOS with an EEPROM programmer.

This is false. For example, you can install Fedora from macOS, but you can not install macOS from Fedora. Does it mean that Fedora put a lock? No, of course. You can install macOS, Windows, or any OS accessing to the BIOS and booting an installer from an USB drive; in other words, you can install any OS following the official way to do it. If you read the Wiki, official way to install HAA is with a wired connection. Installing HAA from Tasmota is a feature provided by Tasmota, and it is under third party installation tools in the Wiki.