RavenSystem/esp-homekit-devices #1325: Support for RS232 controlled devices?

tengelmeier, 2021-02-03:
Hi, 

Sorry if an issue is not the right form to ask:

I'd like to control devices with an RS232 port via Homekit - like an beamer / multimedia receiver. WHere could I start looking to implement it myself? And is this firmware a good start or are the resources (like available memory) already too limited to implement a serial protocol with ~10 accessory features like input selection / on-off / volume / mute / ? 

Thanks!

tonysprenk, 2021-02-03:
Theoretically it’s possible. I have tried this with my Beamer without success though. From my understanding you need a [Rs232 to TTL converter](https://a.aliexpress.com/_uXEC2R) and a [Null modem cable / converter](https://a.aliexpress.com/_v0x6Rt). And obviously you need the (raw) commands. Memory shouldn’t be an issue.

tonysprenk, 2021-02-03:
Let me know if you find some success

closed 2021-03-07 by stale[bot]