RavenSystem/esp-homekit-devices #1326: White only dimmable bulb JSON

ColeDimich, 2021-02-03:
When I try to create a white only dimmable bulb accessory the accessory won't show up to be added in HomeKit. This is the JSON I'm trying to use after reading through the wiki. Am I doing something wrong here?

{
  "c":{"q":850,"b":[{"g":0,"t":5}]},
  "a":[{
    "t":30,
    "w":15,
    “b”:[{“g”:0, “t”:4}]
  }]
}

RavenSystem, 2021-02-03:
Quotes of last line are wrong.

ColeDimich, 2021-02-03:
Thank you for such a quick response! I was diligent about getting the quotes right even though my computer kept wanting to change them, once I got them all correct it added properly. I also wanted to say that I really appreciate the work you've done here and how cool it is that we can create our own HomeKit accessories. Thank you again, all the best

ColeDimich, 2021-02-03:
I also had one more question if it isn't too much of an inconvenience. I haven't been able to get setup mode to work properly after setting up a device. The process is to toggle the button 8 times after plugging the device in correct? When I do that I haven't been able to get the devices wifi network pop up so that the json or wifi network can be updated. Is there something else that needs to be done?

RavenSystem, 2021-02-03:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode#device-connection

closed 2021-03-07 by stale[bot]