RavenSystem/esp-homekit-devices #1327: Updates from old HAA versions is now possible until March 31

RavenSystem, 2021-02-04:
Thanks to @HomeACcessoryKid and GitHub, there is possible to update those devices that were stuck in an old HAA version because changes on GitHub servers.

I have released a special version of HAA Installer to allow updates from old HAA versions installed with LCM v1.0.

Remember to save a copy of your JSON before update, because all saved settings will be removed. HomeKit ID is preserved, meaning that accessory and automations will work again once configured.

rboot4lcm will be replaced with original rboot during update to avoid 1.5s delay at boot introduced with new LCM versions. LCM will be replaced with last version of HAA Installer (aka HAA OTA).

Dead timeline is March 31.

More details:
https://github.com/HomeACcessoryKid/life-cycle-manager/issues/29

axxxkh, 2021-02-06:
hi, i successfully updated 3 devices flashed with ravencore, but i have a problem with updating esp01s (current version haa 1.9.1) i set it up to setup mode and set update haa ota (on setup page it shows update to haa ota 2.2.2.) but after this nothing happened, i can`t visit setup page by ip and don`t shows wifi (like haa04754 etc) and in home devise not responding. when i manually restart device 8 times it come back with the same firmware (haa 1.9.1)

jubudu, 2021-02-21:
Dear all,
I have some hardware installed in walls or other hard to access places with HAA Firmware 0.3.4 and 1.2.1.
Is there a tutorial / guideline how to process the update? Sorry, but I do not find this concrete infos from the links above.
Thank you for your kind support
jubudu

nickforev, 2021-02-22:
> Dear all,
> I have some hardware installed in walls or other hard to access places with HAA Firmware 0.3.4 and 1.2.1.
> Is there a tutorial / guideline how to process the update? Sorry, but I do not find this concrete infos from the links above.
> Thank you for your kind support
> jubudu

Hello. Have you tried 8 times to quickly press the switch to Home?
Then you need to wait 7-12 minutes.

jubudu, 2021-02-22:
Thx, yes I did. I have unmount one to observe the log-file. It stucks during update. May be, a weak WLAN may cause the trouble...

xrust83, 2021-02-27:
Will the RavenCore v1.3.1 firmware update the latest swan or firmware 1.2 without OTA

Przylepzwm, 2021-03-09:
I have many devices in version 0.8.6. Do I have to enter Setup mode in each of them and wait?

closed 2021-03-23 by RavenSystem