RavenSystem/esp-homekit-devices #1329: 4.9.0 and Electrodragon strip controller

sorriso93, 2021-02-07:
I decided to update to 4.9.0 haa due to better wifi management but I get stuck immediately after the json configuration, it seems that HAA doesn't announce it on network. 
Below by uart log, after SAVE on captive portal, and json config. All seems good but I don't see HAA accessory to be added in homekit app.
Any suggestion is welcome, the same json was ok on 3.2

-
Home Accessory Architect v4.9.0
(c) 2019-2020 José Antonio Jiménez Campos

SETUP MODE
Wifi config init
mode : sta(d8:bf:c0:f9:ad:35)
add if0
Saved BSSID: c0c9e39cd086
Wifi Mode: Roaming
Start Wifi smart connect scan
HAA Setup
Hostname: HAA-F9AD35
Enabling auto reboot
mode : sta(d8:bf:c0:f9:ad:35) + softAP(da:bf:c0:f9:ad:35)
add if1
bcn 100
Function called without core lock
Start AP SSID=HAA-F9AD35
bcn 0
del if1
add if1
bcn 100
Start DHCP server
Start HTTP server
scandone
Searching best BSSID for MAXNETHOME
RSSI -57, Ch 1 - c0c9e39cd102
RSSI -51, Ch 1 - c0c9e39cd086
RSSI -67, Ch 1 - c0c9e39cc9a6
Best BSSID: c0c9e39cd086
scandone
reconnect
scandone
add 0
aid 12
cnt 

connected with MAXNETHOME, channel 1
dhcp client start...
ip:192.168.0.86,mask:255.255.255.0,gw:192.168.0.3
bcn 0
del if1
mode : sta(d8:bf:c0:f9:ad:35)

sorriso93, 2021-02-07:
By the way I don't see the electrodragon examples anymore in the post with all tested json. Is there any reason to have it removed?

sorriso93, 2021-02-07:
With this simplified json all ok m. It is strange that with the other even the captive portal is not starting

{"c":{"q":850,"n":"STRCUCTER"},"a":[{"t":30,"r":15,"g":13,"v":12,"w":14,"b":[{"g":0}]}]}

RavenSystem, 2021-02-07:
> By the way I don't see the electrodragon examples anymore in the post with all tested json. Is there any reason to have it removed?

You don't see it because this: 
[image]

sorriso93, 2021-02-07:
> > By the way I don't see the electrodragon examples anymore in the post with all tested json. Is there any reason to have it removed?
> 
> You don't see it because this:
> [image]

You are right I find it

closed 2021-02-18 by sorriso93