RavenSystem/esp-homekit-devices #1332: switch GPIO status always high on boot after V~4.6

djsomi, 2021-02-09:
Hi All,

I've upgraded all HAA devices last week to 4.9 and they started to behave differently. 
All switches are turning ON automatically at boot on if 0 state is not defined.

What changed here? Is there any solution to avoid this? Tried to define "s": 0 in the array but its not working.
Only solution is to define "0" state, but its brings up additional switches in home and this is not ideal in some cases I only want to use only stateless switches.

Thanks!

mariusvasss, 2021-02-09:
Tru this example:

{
  "c":{"l":13,"b":[{"g":0,"t":5}],"0":{"r":[{"g":12}]}},

mariusvasss, 2021-02-09:
After that config, put your JSON

djsomi, 2021-02-10:
For example 3CH sonoff touch wall switch only with stateless buttons:

{"c":{"l":13,"i":0,"b":[{"g":0,"t":5}]},
"a":[
{"t":3,"f0":[{"g":10,"t":1}]},
{"t":3,"f0":[{"g":9,"t":1}]},
{"t":3,"f0":[{"g":0,"t":1}]}
]}

Also I have a sonoff SV in place for garagre door, where I have the same issue, If switch state 0 not defined, relay on connected gpio is always ON when device booting up.

mariusvasss, 2021-02-10:
{"c":{"l":13,"b":[{"g":0,"t":5}],"0":{"r":[{"g":10,"g",9,"g",0}]}},

I think this is it

mariusvasss, 2021-02-10:
And then te 'a' aray

djsomi, 2021-02-10:
Yep, thanks I finally I solved the problem.

I just wanted to know what is changed, and is this an expected operation? I quickly went back on release changelog but I cant see anything related. (or maybe its coming from an older version change, but for me its started with the 4.9 version)

mariusvasss, 2021-02-10:
On the sv just set gpio of relay

RavenSystem, 2021-02-26:
With v5, it is LOW by default:

https://github.com/RavenSystem/esp-homekit-devices/wiki/Accessory-Configuration#Binary-Outputs

closed 2021-02-26 by RavenSystem