RavenSystem/esp-homekit-devices #1334: Sonoff SA-018 & Home Accessory Architect v4.9.0. help

tonton-RISK, 2021-02-10:
Hi, 
need help
Device ---- Sonoff SA-018
Accessory description --- Switch.
Json -----  ?
its on Home Accessory Architect v4.9.0

after switching to Home Accessory Architect v4.9.0 the black wires no longer work.

Thank for your help

[[image]](url)
[[image]](url)
[[image]](url)

tonton-RISK, 2021-02-11:
More picture of SA-018
[[image]](url)

[[URL=https://www.casimages.com/i/210211110533155807.jpg.html][IMG]https://nsa40.casimages.com/img/2021/02/11/210211110533155807.jpg[/IMG][/URL]](url)

[[URL=https://www.casimages.com/i/210211110542653965.jpg.html][IMG]https://nsa40.casimages.com/img/2021/02/11/210211110542653965.jpg[/IMG][/URL]](url)

thanks for help

nickforev, 2021-02-16:
json
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},{"g":4},{"g":4,"t":0}]}]}

tonton-RISK, 2021-02-16:
Hi nickforev,
Thanks for your help , 
Ahan thanks a lot 
I test this 8:00 pm

closed 2021-02-23 by tonton-RISK