RavenSystem/esp-homekit-devices #1335: TV service does not works properly

malvasio, 2021-02-11:
Hello All, I have a IR device, but the only button in the app is power button and SMART-INPUT-GUIDE seccion, no narrows or volume buttons, is anything wrong with the json? 
`
{"c":{"t":14,"p":"U+J=AdAdAdDFAd","o":1},
"a": [
  {"0": {"i": [{"c": "bAeBaFaAcBaAaCb"}]},
  "1": {"i": [{"c": "bAeBaEbAcDaCb"}]},
  "6": {"i": [{"c": "bAeBaEfAaFaA"}]},
  "7": {"i":[{"c": "bAeBaFeAbEaA"}]},
  "8":{"i": [{"c": "bAeBaHhE"}]},
  "9": {"i": [{"c": "bAeBaEaBeAbE"}]},
  "10": {"i": [{"c": "bAeBaEbAcAaBaCaA"}]},
  "11": {"i": [{"c": "bAeBaEcAaAbCaAaB"}]},
  "13": {"i": [{"c": "bAeBaEbAcAaBaCaA"}]},
  "17": {"i": [{"c": "bAeBaEbEaBeA"}]},
  "20": {"i": [{"c": "bAeBaFbAeBaD"}]},
  "21": {"i": [{"c": "bAeBaFbAeBaD"}]},
  "22": {"i": [{"c": "bAeBaEaAfAaF"}]},
  "23": {"i": [{"c": "bAeBaGhF"}]},
  "30": {"i": [{"c": "bAeBaGdAcDaA"}]},
  "q0": [{"h": "192.168.1.58","r": 0}],
  "q1": [{"h": "192.168.1.58"}],"t":60,
  "xa": 0, 
  "i": [{"0": {"i": [{"c":"bAeBaEaAdAaAaDaA"}]},
  "n": "My Apps"},
  {"0":{"i": [{"c":"bAeBaEbEaBeA"}]},"n":"Smart"},
   {"0":{"i": [{"c": "bAeBaGaAfAaD"}]},"n": "Input"},
   {"0":{"i":[{"c":"bAeBaGaAaAaAbAaAaAa"}]},
 "n": "Guide"}]},
{"i":0.5,"1":{"s":[{"a":1}]}}]
}
`

tonysprenk, 2021-02-11:
I believe the other buttons will be displayed in the remote section of the control panel, not in the home app. Buttons in the home app are only for automations.

malvasio, 2021-02-11:
But I have seen others bottons in home app:
[othera](https://user-images.githubusercontent.com/58448955/79885923-81c10b00-8400-11ea-8e4c-46a7e82269bf.png)

tonysprenk, 2021-02-12:
I believe the picture is from control center.

malvasio, 2021-02-15:
> I believe the picture is from control center.

You are right, but there is not volumen bouttoms for example

closed 2021-02-18 by malvasio