RavenSystem/esp-homekit-devices #1336: possible to update from v1.2?

mozzaLB, 2021-02-12:
I have a couple of old switches on RavenCore v1.2. They were controlled by EVE app if I remember right and could be updated through this app. However I do not see the update button in the eve setup anymore. Is it still possible to update them to the latest HAA version?

mozzaLB, 2021-02-12:
is it possible with the LCM update https://github.com/RavenSystem/esp-homekit-devices/issues/1327? It is however not clear to me how to do this. Are there any instructions for noobs like me?

tonysprenk, 2021-02-12:
https://github.com/RavenSystem/esp-homekit-devices/issues/686

mozzaLB, 2021-02-12:
thanks for your response! I also found that solution but the problem is that i do not see FIRMWARE UPDATE in eve setup mode. All the other option are there except that one. Any idea why?

mozzaLB, 2021-02-12:
found it already. I did not install it using LCM. This is the reason i do not see the firmware update option in Eve

tonysprenk, 2021-02-12:
Yes so I believe only option is to flash by connecting to ftdi adapter.

mozzaLB, 2021-02-12:
I also think so. Thanks

closed 2021-03-15 by stale[bot]